ShikOfTheRa/scarab-osd #4: OSD doesn't overlay after flash unless camera is unplugged...

decepts, 2015-03-20:
I recently purchased a RCTimer MACLink-OSD and a MinimOSD v1.1 and I finally flashed both of them with the latest MWOSD v1.3.  Before the MAVLink-OSD had Arducam OSD v2.0 and the MinimOSD had Arducam OSD v2.1.  Before I flashed them both I had plugged them in and checked to see if they are working and I saw them both boot up and go into waiting for heartbeat so I know the display overlay was working great for both.  After flashing the firmware to the latest release I noticed that the OSD appeared to not overlay anything anymore.  After days and weeks of troubleshooting I noticed that if I unplugged the camera IN and left the video OUT the overlay works!  There appears to be some kind of issue with the software.  Reason for this belief is before with Arduino OSD v2.0 I could see the overlay just fine with video but after the flash to MWOSD v1.3 and v1.2 it does not work.  I have read that others have had to use other cameras but other than that I have found nothing.

decepts, 2015-03-20:
Tested with a Sony EFFIO-V WDR Camera and I get OSD and video but not on my other one.

hayden-t, 2015-03-25:
it could be a power problem, how have you got it setup ? so the minim is using power from the 12v video side or not ? if you flash back does the problem go away ? see this page about the 2 power source options: https://code.google.com/p/arducam-osd/wiki/minimosd

ShikOfTheRa, 2015-03-27:
No2 in this link. Almost all time it is incorrect PAL/NTSC setting. Never trust you have been sent the right camera - some think they had NTSC - only to find it was PAL :)

https://github.com/ShikOfTheRa/scarab-osd/blob/master/DOCUMENTATION/FAQ.md

decepts, 2015-04-06:
It is not a PAL or NTSC issue for I verified this.  My setup was hooked up exactly like it is said in that link hayden-t provided.  I now have a LC filter on the 12 volt side mainly for the capacitor because of the power surge from 0 throttle to throttle start up where the OSD would shut down and then come back without overlay and this was corrected by putting in that LC filter which I'm sure if I just have installed a cap it would of been just fine but I already had this laying around so it was quick to install.   If I flash back to Arducam OSD v2.1 everything works great on both cameras but not when I flash it to MWOSD v1.3 or v1.2 where only the more expensive effio-v camera works.  I have not tried any earlier versions of MWOSD yet.

ShikOfTheRa, 2015-04-16:
Have made some changes to SPI and timing.
I suggest to try it with / without vsync. 
Am considering in future to automate this process so that by default just plug in and everything is automatically configured. But with option to manual override.

cyclingengineer, 2015-04-19:
I have this problem with a Turnigy IC-120CS PAL camera. I have overcome it by disabling both AUTOCAM and VSYNC - though the OSD does now flicker a bit and you get sparklies. This is with a minimosd board.

ShikOfTheRa, 2015-04-20:
Thank you for the feedback. Am considering an auto VYSNC for future... if no vsysnc pulse detected it moves into a on vsync mode

cyclingengineer, 2015-04-20:
No problem. Some more information came to light with some googling here - http://www.rc-cam.com/forum/index.php?/topic/4003-minimosd-software-fix-improve-camera-compatibility/

 don't know if your fork includes those changes or not. If not I guess it might be work an investigation.

ShikOfTheRa, 2015-04-20:
Yes that looks similar.  Basically if no vsync, time out and move to non vsync mode...
Maybe I'll rethink having vsync enabled by default until this is in place!! I really want to change as I put a lot of time into improving that code to get the maximum clean screen in fastest time - and happy with the results. Real shame that some cameras are not supporting properly as he advises....

ShikOfTheRa, 2015-04-30:
Autovsync implemented - requires testing

ShikOfTheRa, 2015-04-30:
available for test in beta section...

https://github.com/ShikOfTheRa/scarab-osd/blob/master/DOCUMENTATION/Downloads.md

ShikOfTheRa, 2015-05-05:
Anyone able to test? 
with vysync enabled - doe sit work + also try the autocoam enabled.

cyclingengineer, 2015-05-05:
I may well have some time to test in the next few days.

Thanks!

jbecwar, 2015-05-23:
I was having the same problem with the hobbyking skyzone 200mw kit (http://www.hobbyking.com/hobbyking/store/uh_viewItem.asp?idProduct=65997).  

On 1.3SP2 the OSD would work if the camera was not plugged in.  If the camera video was plugged in after the osd started the osd stopped updating.  

On 1.4beta it works great, right from start up.  Thanks alot of the fix.
-James

closed 2015-06-01 by ShikOfTheRa

trikied, 2015-07-31:
I am having the same issue on R1.4.1
SecurityCam200 camera PZ0420M and mwosd. Before flash it was working fine. If I unplug the camera, I get the screen of OSD showing. are there any fixes for this?

ThoRN2, 2016-01-27:
Hi, had the same problem with my camera.
a.)Video yes. OSD overlay no.
b.)Camera unplugged OSD overlay yes
c.)Plug camera back in while powered. OSD overlay gone

Turns out i recently switched cameras from NTSC to a PAL camera.
1000TVL FPV HD CMOS Camera PAL

Solution 
#define AUTOCAM ---> now OSD overlay and video works, but overlay doesn't line up
set PAL in MwOSD GUI --->  no more problems

Thanks trikied

DasSmeagle, 2016-04-22:
Hi, 

why is this closed, it seems it is not fixed ;) Just flashed my Board from KV_OSD 2.2 to MWOSD R1.6 and lost Overlay. it flickers up for short moments, but most of the time no visible display.

Switching NTSC/PAL and VSYNC did not help.

Will try #define AUTOCAM.

- Oliver

ShikOfTheRa, 2016-04-22:
Because at the moment there is nothing planned that will help this. Other than advise to read the FAQ. 9/10 times it is incorrect PLA/NTSCS

DasSmeagle, 2016-04-22:
Ah, OK... ;)

Switching PAL/NTSC in the GUI just moves the position of the flickering OSD. Guess we continiue talking in #212 

Thanks,
- Oliver

ShikOfTheRa, 2016-04-22:
No it does more than that, Normally running PAL on OSD with an NTSC cam will give no overlay on MWOSD. Or the other way around (I never remember which!). 
Always reboot after changing it....

DasSmeagle, 2016-04-22:
On my System, Toggling between NTSC and PAL in the GUI just changes the position/scaling of OSD... Using a PAL camera. And always had PAL selected at reboot.

ShikOfTheRa, 2016-04-22:
Thats all you see, but it also changes settings on the Maxim chip. So if you change, save and reboot.