ShikOfTheRa/scarab-osd #5: minim osd does not start

redguy666, 2015-03-23:
When I connect minim osd with scarab to multiwii pro (2.3 and 2.4 tested) - it seems it does not start. power leds are on, but yellow led flashes once bright and 2-3 dimmed and then goes off.

when I connect usb device to flight controller - yellow led starts to blink rapidly (as it should). I had very same effect with rush-osd firmware (video: https://www.youtube.com/watch?v=VEhL8eA1d7k)

the main difference is that with scarab I cannot make it work when disconnecting usb (with rush-osd it sometimes worked OK after disconnecting).

I have exactly this minim: http://www.aliexpress.com/item/On-Screen-Display-Ardupilot-Mega-Mini-OSD-Rev-1-1-OSD-diy-drones-APM2-0-APM2/1691027880.html
and this multiwii pro: http://hobbyking.com/hobbyking/store/__26588__multiwii_pro_flight_controller_w_mtk_gps_module.html

now I have multiwii 2.4 and scarab-osd 1.3

any help and suggestions appreciated...

hayden-t, 2015-03-24:
so are you saying it only works with usb ftdi plugged in ?

redguy666, 2015-03-24:
Yes... You can see that on video. Only then yellow light flashes and there is overlay on video... I admit it is strange.
Maybe something is just broken but I dont know which part.

hayden-t, 2015-03-25:
yes strange, but i have a possible theory.  an arduino (atmega chip)  when first powered up checks its serial port for new firmware uploads pending, and if nothing after a short delay it starts the sketch. this is how a ftdi works, it issues a reset to the connected chip and then follows that immediately with the firmware upload.

what i suspect is happening is that you have a actively transmitting multiwii serial port that is confusing your minim into thinking a firmware is attempting to be uploaded to it. 

now the checking and acceptance of a new firmware is done by the boot loader, and so its possible you may have a bootloader on your minim that is not checking the incoming data to see if it is a firmware or not and simply accepting it as being so (but it is not)

the flashing led is generated by the osd sketch, so that is a display if the sketch is running or not.

now the osd can run by itself , even without a active serial connection. so i would suggest you start by making this quick test, disconnect the tx/rx lines going to, so it is getting power only, or power it some other way from a 5v source with no data lines connected and see if it boots and starts flashing, even like this it will put up some osd display data on the screen. 

if this is the case i would suggest you try upgrading your minim's bootloader. you can do this with a avr programmer which connects onto the 6 pin pads that you solder header pins onto where you have one connected with a wire already. use this programmer with arduino ide and the 'burn bootloader' feature

another option would be to look into why you have an active serial port ? im not the sure it should be straight after boot ? maybe you are using it on a serial port chosen in mw config.h to be a gps connection or something else ??

redguy666, 2015-03-26:
Wow... I followed your hints and:
1. connecting only power wires to serial port - did not worked
2. burning bootloader - did not worked...

but.. since only powering it up from serial did not worked - I tried to connect minim to power pins of esc/servo connectors...  and yellow led started... I am going to check voltage on serials as it seems there is no sufficient power and perform further tests. Weird that GPS and bluetooth connected to serials work like a charm :|

redguy666, 2015-03-26:
I can confirm - serials on my MW board provide only 4,1V while on esc/servo pins there is 5V from UBEC. Connecting USB device probably gave additional power source raising it to 5V on serial and making minim osd work fine. As I have separate pins from minim I just will keep it this way - power from esc/servo pins and data from serial.

Thanks for your assistance and helpful suggestions :)

hayden-t, 2015-03-26:
at least you have a work around , maybe your bec are low amperage ?

closed 2015-03-26 by hayden-t