ShikOfTheRa/scarab-osd #6: Won't connect on Ubuntu 14.04 64-bit

cmavr8, 2015-03-25:
The GUI starts, but when I click on the port the FTDI/OSD are connected to, I get a core dump and it crashes.

Using MWOSD 1.3, Ubuntu 14.04 64-bit, Java SE 1.7.0_76.

Parts of the error log:
> #
> # A fatal error has been detected by the Java Runtime Environment:
> #
> #  SIGSEGV (0xb) at pc=0x00007ff9fb58e6e6, pid=9618, tid=140712542066432
> #
> # JRE version: Java(TM) SE Runtime Environment (7.0_76-b13) (build 1.7.0_76-b13)
> # Java VM: Java HotSpot(TM) 64-Bit Server VM (24.76-b04 mixed mode linux-amd64 compressed oops)
> # Problematic frame:
> # C  [librxtxSerial.so+0x76e6]  Java_gnu_io_RXTXPort_nativeDrain+0xd6
> #
> # Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
> #
> # If you would like to submit a bug report, please visit:
> #   http://bugreport.java.com/bugreport/crash.jsp
> # The crash happened outside the Java Virtual Machine in native code.
> # See problematic frame for where to report the bug.
> #

cmavr8, 2015-03-25:
More log:
Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  gnu.io.RXTXPort.nativeDrain(Z)Z+0
j  gnu.io.RXTXPort$SerialOutputStream.flush()V+92
j  processing.serial.Serial.write(I)V+16
j  MW_OSD_GUI.serialize8(I)V+13
j  MW_OSD_GUI.headSerialResponse(ILjava/lang/Boolean;I)V+3
j  MW_OSD_GUI.headSerialReply(II)V+21
j  MW_OSD_GUI.SendCommand(I)V+498
j  MW_OSD_GUI.InitSerial(F)V+166
j  MW_OSD_GUI.controlEvent(LcontrolP5/ControlEvent;)V+25
j  sun.reflect.GeneratedMethodAccessor1.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+
40
J 418 C2 java.lang.reflect.Method.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object; (63 by
tes) @ 0x00007ffa4917ab70 [0x00007ffa4917aac0+0xb0]
j  controlP5.ControlBroadcaster.invokeMethod(Ljava/lang/Object;Ljava/lang/reflect/Method;[Ljava/lang/Obj
ect;)V+22
j  controlP5.ControlBroadcaster.broadcast(LcontrolP5/ControlEvent;I)LcontrolP5/ControlBroadcaster;+247
j  controlP5.ListBox.controlEvent(LcontrolP5/ControlEvent;)V+104
j  controlP5.Controller.broadcast(I)V+47
j  controlP5.Button.setValue(F)LcontrolP5/Button;+7
j  controlP5.Button.activate()V+48
j  controlP5.Button.mouseReleased()V+14
j  controlP5.Controller.setMousePressed(Z)Z+114
j  controlP5.ControllerGroup.setMousePressed(Z)Z+32
j  controlP5.ControllerGroup.setMousePressed(Z)Z+32
j  controlP5.ControlWindow.mouseReleasedEvent()V+39
j  controlP5.ControlWindow.mouseEvent(Lprocessing/event/MouseEvent;)V+44
j  sun.reflect.GeneratedMethodAccessor2.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+
40
J 418 C2 java.lang.reflect.Method.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object; (63 by
tes) @ 0x00007ffa4917ab70 [0x00007ffa4917aac0+0xb0]
j  processing.core.PApplet$RegisteredMethods.handle([Ljava/lang/Object;)V+23
j  processing.core.PApplet.handleMethods(Ljava/lang/String;[Ljava/lang/Object;)V+18
j  processing.core.PApplet.handleMouseEvent(Lprocessing/event/MouseEvent;)V+192
j  processing.core.PApplet.dequeueEvents()V+53
j  processing.core.PApplet.handleDraw()V+301
j  processing.core.PGraphicsJava2D.requestDraw()V+4
j  processing.core.PApplet.run()V+179
j  java.lang.Thread.run()V+11
v  ~StubRoutines::call_stub

hayden-t, 2015-03-25:
hi, can you try jre 1.7 32bit . i found problems with x64 and the java gui's

cmavr8, 2015-03-25:
Hi! thanks for the reply!
Seems to work, but I don't have the OSD/FTDI with me right now! I'll try in the evening and report back.

hayden-t, 2015-03-25:
great to hear that :)

ShikOfTheRa, 2015-03-27:
Good to know...

cmavr8, 2015-03-28:
Still, this is annoying, as I have to keep switching (using $ sudo update-alternatives --config java) between the default and the 32-bit version of java. Arduino won't work with 32-bit, and MW_OSD_GUI won't work with 64.

I tried setting the environmental variable JAVA_HOME to the dir of the 32-bit installation, hoping that MW_OSD_GUI would pick it up, but it doesn't :(

btreecat, 2015-03-31:
Can confirm the same issues in Arch linux 64bit. I tried installing 32bit JVM but ran into the same issues. Currently my only option is to boot into windows so I would love to see some progress on this. Any way I can help track down the issue?

cmavr8, 2015-04-01:
Kind of off topic here, but what is the problem with the 32-bit version of Java? I had to download the tar, place it in the right directory, then run update-alternatives like this:
sudo update-alternatives --install /usr/bin/java java /usr/lib/jvm/jre1.7.0_76/bin/java 4
where the 4 is the first free "slot" available, when running sudo update-alternatives --config java. So I was getting 3 options so I added the 4. 

Now I run sudo update-alternatives --config java and select 4 to run MW_GUI, or 0 to run all my other/normal Java programs. It's cumbersome, but it works.

But yes we should track it down properly.
The error I get when running MW_OSD_GUI 64-bit version is 
hudlayout.xml does not exist or could not be read

ShikOfTheRa, 2015-04-02:
hudlayout.xml should exist in processing data folder. This shoudl be default folder for processing app might be worth checking to see if any capitalisation errors? / code looks for all lowercasename

cmavr8, 2015-04-02:
I don't understand your comment entirely. But hudlayout.xml exists in MW_OSD_GUI/application.linux64/data, and is lowercase as it should.

I also tried giving the script (MW_OSD_GUI/application.linux64/MW_OSD_GUI) the full java path (of the 32-bit java) but it won't work, I get "wrong ELF class".

btreecat, 2015-04-02:
OK I was able to get it to work on Arch Linux with 32bit java. I realize now that the script was calling the 64bit java still so once I forced it to the 32bit run-time it worked fine (until windows killed my FTDI chip which is now working again =D)

cmavr8, 2015-04-05:
Could you please give us tips on how you did it?

ShikOfTheRa, 2015-04-08:
Wish I could help guys - have no linux knowledge at all.
maybe could install processing and run the source code directly??

sounds like btreecat may have the answers though...

btreecat, 2015-04-08:
I'll be glad to give this a try, just hadn't had time. Glad to know
processing is a req, should be easy to get my environment set up.

@shikofthera no worries, we all started somewhere :-)

ShikOfTheRa, 2015-04-16:
Guys - am going to close this as I can't support linux. 
MAc is just as bad... lol windoze all the way!

btreecat, 2015-04-16:
@shikofthera That is not an acceptable answer.

I don't mind helping you support linux, and you can easily install it on a
spare partition/hard drive. I can even help you with that if you would
like.

I'll take a look at this bug this weekend and see what I can find.

@shikofthera what are the typical steps to setup your build environment?
I'd like to try and replicate it on my system with as little guesswork as
possible :-)

cmavr8, 2015-04-16:
I propose we leave it open. btreecat or me may find the time/courage to debug it at some point.
Or someone else who comes in may do so.
It's ok to leave it long running and open. Maybe one of the labels can be assigned.

ShikOfTheRa, 2015-04-16:
Ok - no issues here.

ShikOfTheRa, 2015-04-16:
@btreecat - its best answer I can give:) Its simply a case of prioritising my time. I just can't give it to ensuring compatibility with all different flavours of linux/mac/pc etc... hardly have time to focus on the core frequently used stuff.
I don't do anything for special linux - simply press teh box that says produce the linux export! mac doesn't seem to work either...

If you guys find anything that needs changing then left me know - all I really do it tick the box taht says produce.

Someone is intending to replace the gui with a chrome version same as baseflight/cleanflight config.
I can't wait because I absolutely hate the current GUI, what is programmed in and how its been implemented. 

To implement:
Install processing from processing.org
Then "download windows zip" from https://github.com/ShikOfTheRa/scarab-osd
browse to gui folder and double on the MWOSD gui file
then press the play icon.

btreecat, 2015-04-16:
Cool thanks for the info @shikofra !

It is probably some silly bug somewhere, I am sure Chris or I can track it
down before the new UI is available.

I would be interested to find out who is working on a chrome app as I have
had great success using cleanflight in various flavors of Linux.

Also off topic, have you had time to work on cleanflight 1.8 API support?
There were some changes and a few things no longer save through the OSD
menu.

cmavr8, 2015-04-16:
Yes it's absolutely understandable that a dev does not have enough time/motivation to debug every platform or implement every requested feature. This is open source after all, no strings attached.

Cheers all

ShikOfTheRa, 2015-04-16:
For cleanflight specific support it will be limited to probably just getting the RC stuff working again. Unfortunatelythe recent additions to seperate pirch / roll rates broke isn't compatible
There is no memory spare to do much. Can't even run with all features enabled. 
Every time need to add something new it means optimising memory somewhere else.Takes time

fnurgel, 2015-04-25:
I'm running same Ubuntu version and got the same problem described here.
I downloaded the Rxtx2.2pre2 source and compiled it myself (after modding it to actually compile) and now I can run the GUI on 64-bit java just fine.

btreecat, 2015-04-25:
Great work @fnurgel!

So does this mean the source version just needs updated and recompiled in
order to distribute the fix?

fnurgel, 2015-04-25:
Actually, the one in "Upgraded RXTX/x86_64-unknown-linux-gnu/librxtxSerial.so" from this repo also seems to work.
The one in the dist is an older one.

btreecat, 2015-04-25:
Maybe the issues with OS X and Linux is just needing to compile it on a
*nix platform. I don't mind setting up Jenkins on my VPS and have it auto
pull/compile binaries.

OTOH, since it is a different version, maybe @shikofthera just needs to
release an update with the new lib?

@shikofthera would you mind compiling it for linux again and sharing it
with us to test?

ShikOfTheRa, 2015-04-25:
sounds promising... try this gui zip file...

ShikOfTheRa, 2015-04-26:
helps if send link...
https://drive.google.com/folderview?id=0B2MInRUrbpWxSk9hZ3JIRXNRUU0&usp=sharing

fnurgel, 2015-04-28:
That gui zip works for me on 64-bit

ShikOfTheRa, 2015-04-28:
OK. thats good feedback. So all I did was updated the file in the processing compiler. Strange thing is I thought I had done this before.... so now to test a mac version

ShikOfTheRa, 2015-04-30:
Included in latest pre-release. Hopefully doesn't break others...

kaefert, 2015-05-09:
hi there! just wanted to share that I hit the same issue while using from your Download-page the "MW OSD R1.3SP2" release. - Then I found this conversation here, tried the "LATEST COMPILED BETA TEST RELEASE" = "MWOSD R1.4 PRE 1.3" and that worked.

one additional thing I thought I'd mention somewhere: It took me a while to find that I needed to press the reset button of my MinimOSD just before installing a new software on it, otherwise it wont work. probably this is because my serial adapter does not have a line called RESET, GRN, or DTR - I only have connected power, TX and RX.

closed 2015-06-01 by ShikOfTheRa