ShikOfTheRa/scarab-osd #7: Crazy idea: MAVlink support?

cmavr8, 2015-03-25:
Seeing how active, properly setup and developed this project is, I can't help but wonder why equivalent project for ardupilot is so underdeveloped.

Taking into account that the whole ardupilot community has to rely on the MinimOSD-extra project, it's painful to watch how slowly that moves, and how behind it has fallen. I don't mean to criticize the developers, they're doing their best and we all only have so much time left after our day jobs. I'm just saying that MW OSD's progress is impressive.

So, I was thinking, how hard would it be for MW OSD to support parsing of the MAVlink protocol, so that it can be used on APM/Pixhawk setups, too? I guess this is the relevant code https://github.com/diydrones/MinimOSD-Extra/blob/master/MinimOsd-Extra/MAVLink.ino

This is not a feature request, just a discussion.

hayden-t, 2015-03-25:
I think it would be pretty easy and something our main developer shikra has mentioned to me too: > " today its called mwosd - mw being multiwii, but maybe we have to consider changing this in future - now supports baseflight etc + standalone gps. I would like to put ardupilot in there too."

OshCFII, 2015-03-25:
I run both a Pixhawk (ardupilot) as well as Cleanflight.   It would be great to run the same OSD on both platforms.

ShikOfTheRa, 2015-03-27:
Yes - I would like to do this. Possibly 1.5
It just needs some time to find out the info. 

If someone else has time to do that would be even better:)

ShikOfTheRa, 2015-07-20:
Guys I am going to close this - its on the future roadmap so no need to duplicate

closed 2015-07-20 by ShikOfTheRa