ShikOfTheRa/scarab-osd #9: RC rates not saving under cleanflight .63.

rgury, 2015-04-03:


VirtualEnder, 2015-04-29:
The serial protocol was updated in cleanflight 1.8.X to separate out roll and pitch rates.  Nothing is saving at all on an of the rates in MWOSD now as a result of the change.

ShikOfTheRa, 2015-04-29:
Yes - this is next item on my list! Will be looking to implement an interim solution for this in next couple of days. At least until Supersopp or me have time to implement a more elegant solution.

ShikOfTheRa, 2015-04-30:
Interim solution in here for test
https://github.com/ShikOfTheRa/scarab-osd/blob/master/DOCUMENTATION/Downloads.md

Arakon, 2015-05-05:
Using R1.4 Pre 1.1, saving of  RC rates still doesn't work. PIDs work just fine, however.

ShikOfTheRa, 2015-05-05:
Hello - are you able to test R1.4 Pre 1.2 for me?. Sorry for hassle but I don't have a working config of cleanflight at the moment. If this doesn't work I'll convert my copter to test.

Download in here:
https://github.com/ShikOfTheRa/scarab-osd/blob/master/DOCUMENTATION/Downloads.md
Only really need to update the OSD sketch, but there are also a few GUI changes / improvements as per the feedback I read this morning.

ShikOfTheRa, 2015-05-05:
Thanks mate - typo there well spotted!! Hate keep having to ask someone to test. Sooner I get a working cf install again the better.

The original issue was I didnt realise CF checked to ensure there was 10 bytes in MSP. Didn't know that. I guess BF does the same too. 

Updated the repo and zip build for testers. Hopefully before anyone realised!

Arakon, 2015-05-05:
Working now, just gave it a try.

ShikOfTheRa, 2015-05-05:
brill - will leave open for a bit just in case anything else needed

ShikOfTheRa, 2015-05-06:
Now updated and with better stuff from supersopp

closed 2015-05-06 by ShikOfTheRa