ShikOfTheRa/scarab-osd #11: standalone GPS error.

slappomatt, 2015-04-10:
Mentioned this to Shikra before on the lab. 

Arduino: 1.0.6 (Windows 7), Board: "Arduino Nano w/ ATmega328"
GPS.ino: In function 'void SerialGpsPrint(char)':
GPS:38: error: 'str' was not declared in this scope
GPS.ino: In function 'void GPS_SerialInit()':
GPS:74: error: invalid conversion from 'char*' to 'char'
GPS:74: error: initializing argument 1 of 'void SerialGpsPrint(char)' 

Was my error msg.

ShikOfTheRa, 2015-04-10:
Fixed in dev pending test.

slappomatt, 2015-04-10:
I love you.

ShikOfTheRa, 2015-04-15:
Feeling the love. 
I'm guessing you aint females, 18, fit  with blonde hair and big tits?

slappomatt, 2015-04-15:
As a matter of fact.... I will be in a few months... <3

ShikOfTheRa, 2015-04-16:
It compiles in the last release. I also ran a quick test flight with GPS emulator and it all seemed OK.

ShikOfTheRa, 2015-04-16:
Closing ! Feel free to send a naked picture of yourself once you turn into that female a8 blonde chick.... not before....

closed 2015-04-16 by ShikOfTheRa