ShikOfTheRa/scarab-osd #21: R1.3 (all) - Cannot edit font

dale3h, 2015-04-22:
## Computer 1 ##
Windows 7 Ultimate (64-bit)
Java 7 Update 13 (32-bit, build 1.7.0_13-b20)
GUI: R1.3, R1.3 SP0, R1.3 SP1, R1.3 SP2 (Windows 32-bit)

* Load font "large.mcm"
* Edit char 0x10
* Save char
* Entire font reverts back to "default.mcm"

## Computer 2 ##
Windows 7 Ultimate (64-bit)
Java 7 Update 71 (64-bit, build 1.7.0_70-b14)
GUI: R1.3, R1.3 SP0, R1.3 SP1, R1.3 SP2 (Windows 64-bit)

* Load font "large.mcm"
* Edit char 0x10
* Save char
* Entire font reverts back to "default.mcm"

ShikOfTheRa, 2015-04-23:
This is not well documented, but from memory it creates a file called custom.mcm from memory. You then have to load that file. This was to protect the defaults as they were getting accidentally messed up.

dale3h, 2015-04-23:
Ok, that seems to work a little better, but every time I click "Save" it still reverts back to "default.mcm" and I have to go reload "custom.mcm". Is there any way around this? I'm a compulsive saver >_<

ShikOfTheRa, 2015-04-23:
Yep - forget built in editor - better to use this one in the repository...
https://github.com/ShikOfTheRa/scarab-osd/tree/master/MW_OSD_FONTS

dale3h, 2015-04-23:
That's what I resorted to trying :) I do not need to edit the fonts personally, but a user of my MinimOSD Flashpack ran into the issue and I told him I'd forward the response to you.

Is there any way to integrate this into the GUI so that we can prevent someone else from getting frustrated?

ShikOfTheRa, 2015-04-24:
Could do. As long as I don't upset the original creator :)
Can't please everyone as I was requested to stop it overwriting the defaults !!

ShikOfTheRa, 2015-04-24:
On second thoughts - if I change the link I have no idea if it breaks linux / mac gui. If I can get that confirmed I'll change. If not will have to stay as is.

dale3h, 2015-04-24:
I can test it on Mac OS X (Yosemite). I also have someone else who can test it on Linux (Ubuntu).

ShikOfTheRa, 2015-04-24:
cool - let me know if both work - and I'll amend

dale3h, 2015-04-24:
So you just want me to test the MAX7456Charwizard.jar standalone and let you know if it works on both platforms? Or did you want to change the link, send me the binaries, and test it that way?

ShikOfTheRa, 2015-04-24:
just see if it loads! not sure if I know how to run from within the gui yet, but lets figure that out if its possible

ShikOfTheRa, 2015-05-05:
So it seems to workish in mac - as long as put editor file in data folder.
Am inclined to just bundle it included for now and not to set up links as concerned at wider scale impacts.
Will put on wish list for a one day might do it....

closed 2015-06-01 by ShikOfTheRa