ShikOfTheRa/scarab-osd #33: Clock running slow

VirtualEnder, 2015-05-02:
I just noticed this by accident when I was editing some video from my DVR.  The timer is runing significantly slow. From arm to disarm in the video is just over 4 minutes, and the OSD is showing 3:31.  It gets even more pronounced on longer runs.  After noticing I started paying attention, and a flight today the OSD said 5 minutes and the timer on my transmitter said almost 7.

ShikOfTheRa, 2015-05-03:
Hopefully realised in 1.4 
Issue has been there day one, but I think recent improvements and changes to sync code has highlighted the problem much more.

therippa, 2015-05-03:
I'm having this problem too, it's telling me ~2:00 after flights that are ~5:00.

VirtualEnder, 2015-05-05:
I just uploaded 1.4 r1.1 and it seems to have solved the problem. Thanks!

therippa, 2015-05-05:
Where are you finding 1.4 r1.1?  I'm not seeing that branch/tag.

ShikOfTheRa, 2015-05-05:
HAd 2 reports this is fixed - so closing. Thanks...

closed 2015-05-05 by ShikOfTheRa