ShikOfTheRa/scarab-osd #42: 32-bit: MW_OSD_GUI.exe crashes with "Unspecified Error"

chrisinvb, 2015-05-08:
When double clicking on MW_OSD_GUI.exe inside of the /application.windows32/, the following error message appears:

"Unspecified Error"

[image]

There is no "C:\ProgramData\Oracle\Java\javapath\javaw.exe" path on my computer.

chrisinvb, 2015-05-08:
To get the executable to run, I copied the MW_OSD_GUI.bat from the /application.windows64/ into the /application.windows32/ folder and ran the MW_OSD_GUI.bat so that the correct java libraries would load.

VirtualEnder, 2015-05-12:
This is a pretty common error with Processing code.  It has to do with a change in the way oracle handles updating the local $PATH system variable.  Adding the non-aliased path (the path showing there is a shortcut, not the actual path) to your java installation also solves the problem.

ShikOfTheRa, 2015-06-01:
Added to FAQ #20
https://github.com/ShikOfTheRa/scarab-osd/blob/master/DOCUMENTATION/FAQ.md

closed 2015-06-01 by ShikOfTheRa