ShikOfTheRa/scarab-osd #43: MW_OSD_GUI.bat throws exception for MW_OSD_GUI/application.windows64

chrisinvb, 2015-05-08:
When executing MW_OSD_GUI.bat (MWOSD R1.3 SP2), the following exception is thrown:

ControlP5 2.0.4 infos, comments, questions at http://www.sojamo.de/libraries/con
trolP5
java.lang.UnsatisfiedLinkError: C:\Users\DELL-E6510\Google Drive\Engineering Tec
h\OSD\MWOSD\MW_OSD_GUI\application.windows64\lib\rxtxSerial.dll: Can't load AMD
64-bit .dll on a IA 32-bit platform thrown while loading gnu.io.RXTXCommDriver
java.lang.UnsatisfiedLinkError: C:\Users\DELL-E6510\Google Drive\Engineering Tec
h\OSD\MWOSD\MW_OSD_GUI\application.windows64\lib\rxtxSerial.dll: Can't load AMD
64-bit .dll on a IA 32-bit platform
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(Unknown Source)
        at java.lang.ClassLoader.loadLibrary(Unknown Source)
        at java.lang.Runtime.loadLibrary0(Unknown Source)
        at java.lang.System.loadLibrary(Unknown Source)
        at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:123)
        at processing.serial.Serial.list(Unknown Source)
        at MW_OSD_GUI.setup(MW_OSD_GUI.java:804)
        at processing.core.PApplet.handleDraw(PApplet.java:2280)
        at processing.core.PGraphicsJava2D.requestDraw(PGraphicsJava2D.java:243)

        at processing.core.PApplet.run(PApplet.java:2176)
        at java.lang.Thread.run(Unknown Source)
Exception in thread "Animation Thread" java.lang.RuntimeException: Error inside
Serial.ports()
        at processing.serial.Serial.errorMessage(Unknown Source)
        at processing.serial.Serial.list(Unknown Source)
        at MW_OSD_GUI.setup(MW_OSD_GUI.java:804)
        at processing.core.PApplet.handleDraw(PApplet.java:2280)
        at processing.core.PGraphicsJava2D.requestDraw(PGraphicsJava2D.java:243)

        at processing.core.PApplet.run(PApplet.java:2176)
        at java.lang.Thread.run(Unknown Source)

chrisinvb, 2015-05-08:
It seems that the executable is not in the /application.windows64/ directory.  There is only the batch file.

ShikOfTheRa, 2015-05-08:
I am getting the same.... will investigate

ShikOfTheRa, 2015-05-09:
Mm looks like a Java update this week is causing this.

ShikOfTheRa, 2015-05-11:
As an update - all previous versions of the GUI will no longer operate for me in 64 bit mode. 32 bit all work fine.
It is same for multiwii GUI.

So I guess my conclusion is the Java update last week appears to have changed something!
On my PC I now seem to have 32 bit version of Java which is interesting. I'm sure it was 64 bit before update!
Suggest accept 32bit is OK - or if concerned install the 64 bit version of java

As the 32 bit version works I am not too concerned as long as people have an option that works - there is no benefit from running 64 or 32 bit GUI. As long as there is one operating one :)

ShikOfTheRa, 2015-06-01:
FAQ # 20 may help:
https://github.com/ShikOfTheRa/scarab-osd/blob/master/DOCUMENTATION/FAQ.md

closed 2015-06-01 by ShikOfTheRa