ShikOfTheRa/scarab-osd #53: Crosshair at screen center

CrazyCoder, 2015-05-19:
If you enable horizon, you can see a nice crosshair, it would be great to have only crosshair, but not the artificial horizon. Maybe it's already possible?

I workaround would be to add CallSign like "+" and move it to the center, but CallSign is always flashing which is annoying.

Maybe there is a better way that I'm not aware of? Thanks!

ShikOfTheRa, 2015-05-19:
Yes this is fine. 

Added but not tested yet..
//#define FORCECROSSHAIR            // Forces a crosshair even if no AHI / horizon used

CrazyCoder, 2015-05-19:
Thanks, I'll give it a try once this commit is shown by github. Any chance it could be added as an element that you can add/remove via layout?

This way you will be able to enable/disable it with a switch.

ShikOfTheRa, 2015-05-19:
OK - will have to look at this in a little more detail - as I think the AHI stuff is linked to haveing an accelerometer- which I guess you may not have. Will try to include it as layout option.

At a guess - cross hair to aim and fire?? :)

CrazyCoder, 2015-05-19:
To aim at the race gates and tight spots, mainly.

I found in config.h that CallSign blinking can be disabled so that it's permanently displayed. This provides a hack to use it at the screen center with some symbol as a crosshair.

Of course it would be really cool to have it configurable via GUI or menu, so this request still remains valid.

ShikOfTheRa, 2015-05-20:
will be a few days as am out office

ShikOfTheRa, 2015-05-22:
Hello - latest repository contains updated compiled version with a few enhancements and all logged bugs fixed (hopefully) This is R1.4 pre-release no 4. Hopefully the last :)

available from link at bottom of http://www.mwosd.com

Of note:
Includes an untested option to support Cleanflight 1.9
Fixes for GPOSD incorrect sat data
A number of small recent requests - see list of changes below

Full features
https://github.com/ShikOfTheRa/scarab-osd/blob/master/DOCUMENTATION/Features.md

CrazyCoder, 2015-05-22:
Thanks, I've already tested the always on crosshair without AIH, it works great, but is present on all layouts. I wanted it to be configurable to have a layout without crosshair.

Also already built 1.4 pre-release from Git and installed it on my Micro MinimOSD, seems to work with Cleanflight 1.9.

ShikOfTheRa, 2015-05-25:
At moment its tied to "horizonPosition" on layout editor. If you are not using sidebars / ahi, this will work to enable / diasble. If not - 15. can be added.

closed 2015-06-01 by ShikOfTheRa