ShikOfTheRa/scarab-osd #57: Cleanflight 1.9 RC TUNING labels are wrong

CrazyCoder, 2015-05-24:
Yaw and Pitch rates are now separate, so all the parameters after ROLL PITCH RATE are off.

YAW RATE displays PITCH RATE.
TPA displays YAW RATE.
THROTTLE MID displays TPA
etc...

Also,

ROLL displays 0
PITCH displays TPA Breakpoint value.

CrazyCoder, 2015-05-25:
This problem is a result of removing `CLEANFLIGHT` define and replacing it with `CLEANFLIGHT190`, `CLEANFLIGHT181` and `CLEANFLIGHT180`. The code in `GlobalVariables.h` still depends on `CLEANFLIGHT` define which no longer exists.

ShikOfTheRa, 2015-05-25:
Yes - thank you for the feedback. Unfortunately I don't have a setup to test cleanflight so relaying entirely on community testing at the moment. Have made those changes

ShikOfTheRa, 2015-06-01:
Hello - any chance to test beta download PRE1.5 is working OK ?

CrazyCoder, 2015-06-01:
Should work, I've compiled it after a fix and it worked fine.

ShikOfTheRa, 2015-06-01:
OK - thanks for quick reply!. See if I can get a few other testers to OK the cleanflight then post the latest compiled as full release.

ShikOfTheRa, 2015-06-19:
OK no bad feedback so far from latest. Closing  :)

Thanks for assistance..!!

closed 2015-06-19 by ShikOfTheRa