ShikOfTheRa/scarab-osd #63: Issue changing battery cells to 3 in configurator

grahamwho, 2015-05-30:
I'm using 1.4pre5 and the x64 configurator  and when I go to change the battery cells from the default of 4 to 3 I keep having this issue when trying to write where it just says waiting for FC forever.  I've tried changing it to 6 and also back to 4 and it works fine with either of those values but it won't write if I have it on 3 battery cells. I've also found that if i deselect display battery voltage and have it on 3 cells that will work.   Strange issue, has anyone else seen this before?
Do I even need to select the right battery cells since i'm using cleanflight and getting my battery info from the flight controller?

ShikOfTheRa, 2015-06-01:
Unable to replicate this. And I can't see any reason why its only like this for 3! It can happen if OSD remains connected to the flight controller when updating. - but should affect other values too

Its only required if use the battery status icon - "display battery status" enabled on gui.
If desperate - I think this can be changed via OSD menu?

closed 2015-06-01 by ShikOfTheRa

grahamwho, 2015-06-21:
So I tried this again, and it's anything below 3 not just 3.