ShikOfTheRa/scarab-osd #68: GPS altitude reset at arm not working

misakm, 2015-06-13:
Hello,

I have the lastest version R 1.4 and cleanflight 1.9.0. When I arm my quadrocopter GPS altitude is not reset. It is not disabled in config file - //#define DISABLEGPSALTITUDERESET 

Thank you for any tip how to resolve this issue.

BR, Michal

ShikOfTheRa, 2015-06-15:
Its only used in fixedwing.
You can add for quad by adding following line into config.sys
  #define RESETGPSALTITUDEATARM

Note- the default altitude from flight controller is BARO altitude.

misakm, 2015-06-16:
Thank you. I will test it. I need GPS altitude because of problem with baro on my flip32. The sensor is crazy.

ShikOfTheRa, 2015-06-16:
OK, I did wonder! Unusual to have GPS and no BARO - instead suggest maybe try:
#define FIXEDWING
#define USEMAGHEADING
effectively this replaces all baro stuff with gps...

ShikOfTheRa, 2015-06-19:
As fyi - have made this a default option - with ability to disable if required. I think this suits the majority of people best.

closed 2015-06-19 by ShikOfTheRa

misakm, 2015-06-23:
I havent time yet to test the settings. Do you mean that I should use
#define RESETGPSALTITUDEATARM and also #define FIXEDWING,  #define USEMAGHEADING.
Can you please describe FIXEDWING and USEMAGHEADING parameters?

ShikOfTheRa, 2015-06-23:
All it does is relpace baro parameters with GPS ones. Also renames acro to gyro as its more known to fixed wing flyers. I think thats all