ShikOfTheRa/scarab-osd #70: Fails to store with voltage alarm set to 9.9

dustin, 2015-06-18:
Configuring 3S with a voltage alarm set to 9.9V, saving just gets stuck on "Waiting for FC".  Setting to 10V works fine.

ShikOfTheRa, 2015-06-19:
What version? 1.4 ?

Really weird!!
I am unable to replicate this with 1.4

Also just wiped and installed fresh defaults to be sure.

dustin, 2015-06-19:
Yes, 1.4 release.  I tried several times with this saved configuration on an OSDoge.  I'd load the config, change voltage alarm to 3.3, hit save and it'd hang on "Waiting for FC"

(can't attach the actual file here, so here come the contents):

```xml
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE properties SYSTEM "http://java.sun.com/dtd/properties.dtd">
<properties>
<comment>MWOSD Configuration File  Wed Jun 17 23:57:24 PDT 2015</comment>
<entry key="S_CS2">54.0</entry>
<entry key="S_CS1">71.0</entry>
<entry key=" - GPS Altitude">0.0</entry>
<entry key="S_CS0">75.0</entry>
<entry key="Speed alarm">100.0</entry>
<entry key="RC Switch">1.0</entry>
<entry key="Voltage Alarm">10.099999</entry>
<entry key="Battery Cells">3.0</entry>
<entry key="Timer alarm">30.0</entry>
<entry key=" - Map mode">1.0</entry>
<entry key="Units">0.0</entry>
<entry key="Time Zone offset">0.0</entry>
<entry key="Amperage 16L">150.0</entry>
<entry key="Debug">0.0</entry>
<entry key="Amperage 16H">0.0</entry>
<entry key="RC Switch ch">5.0</entry>
<entry key="Use FC RSSI">1.0</entry>
<entry key="Display Gimbal">0.0</entry>
<entry key="x10   Altitude alarm">100.0</entry>
<entry key=" - SB direction">1.0</entry>
<entry key="Time Zone +/-">0.0</entry>
<entry key="Use FC video voltage">0.0</entry>
<entry key=" - SB Scrolling">1.0</entry>
<entry key="Use BoxNames">0.0</entry>
<entry key="Display Vario">0.0</entry>
<entry key="Display Horizon Bar">1.0</entry>
<entry key=" - Coords on Top">1.0</entry>
<entry key="Display Video Voltage">0.0</entry>
<entry key="HUD layout High">10.0</entry>
<entry key="Use Virtual Sensor">0.0</entry>
<entry key="Display CallSign">1.0</entry>
<entry key="Video Signal">0.0</entry>
<entry key="Voltage Adjust">200.0</entry>
<entry key="Display Compass">0.0</entry>
<entry key="RSSI Alarm">60.0</entry>
<entry key="Use FC amperage">1.0</entry>
<entry key="Display BARO ALT">1.0</entry>
<entry key="Display Voltage">1.0</entry>
<entry key="Display Amps">1.0</entry>
<entry key="RSSI Min">0.0</entry>
<entry key="Use PWM">0.0</entry>
<entry key=" - FM sensors">1.0</entry>
<entry key="Reset Stats After Arm">0.0</entry>
<entry key="x100 Distance alarm">100.0</entry>
<entry key="Display Heading">0.0</entry>
<entry key="Display Angle to Home">0.0</entry>
<entry key="Zero Adjust">4.0</entry>
<entry key="Display GPS time">0.0</entry>
<entry key="Use FC main voltage">1.0</entry>
<entry key="Display RSSI">1.0</entry>
<entry key="EEPROM Loaded">1.0</entry>
<entry key="x100 mAh Alarm">9.0</entry>
<entry key="Enable ADC 5v ref">0.0</entry>
<entry key="Display Side Bars">1.0</entry>
<entry key=" - GPS Coords">1.0</entry>
<entry key="Display Battery Status">1.0</entry>
<entry key="Display Throttle Position">1.0</entry>
<entry key="Display Timer">1.0</entry>
<entry key="RSSI Max">150.0</entry>
<entry key=" - Heading 360">1.0</entry>
<entry key="HUD layout Low">0.0</entry>
<entry key="Display GPS">1.0</entry>
<entry key=" - HB Elevation">0.0</entry>
<entry key="Display Flight Mode">1.0</entry>
<entry key="Amp Alarm">35.0</entry>
<entry key="HUD layout Mid">1.0</entry>
<entry key="Display mAh">1.0</entry>
<entry key="Amps Adjust">150.0</entry>
<entry key="S_CS9">0.0</entry>
<entry key="S_CS8">0.0</entry>
<entry key="S_CS7">0.0</entry>
<entry key="S_CS6">0.0</entry>
<entry key="S_CS5">70.0</entry>
<entry key="S_CS4">87.0</entry>
<entry key="S_CS3">72.0</entry>
</properties>
```

ShikOfTheRa, 2015-06-19:
mm. works for me.
what is odd is the  entry : <entry key="Voltage Alarm">10.099999</entry>
I can load it fine - but its not a vlaue that should be ther eunless from an older pre-release gui
There was an issue with one of the 1.4 beta test releases a while back

maybe - try ...
#1 changing to 10.0 in the file
#2 make sure download and use gui from R1.4.1

ShikOfTheRa, 2015-07-10:
I could never replicate - will leave open for a while and if OK will close in  week  or so

dustin, 2015-07-13:
Still happens to me, but if it's just my config, I'll deal with it.  :)

ShikOfTheRa, 2015-07-13:
Its weird - suggest:

#1 changing to 10.0 in the file
#2 use gui from latest release

ShikOfTheRa, 2015-07-20:
OK - still unable to replicate, but I did notice that at one point, with one of the dev releases it would have save din the format you mentioned. I think it could be real bad luck that you had that whilst that was the latest in repo. Was only there for a v short time.

I'll close for now!

closed 2015-07-20 by ShikOfTheRa