ShikOfTheRa/scarab-osd #84: How do I run the simulator?

u8mydog, 2015-07-03:
I can't figure out how to run the simulator. It would be nice if this info was captured in the documentation.

u8mydog, 2015-07-03:
Looks like I got it working. But still would be nice to have the documentation.

ShikOfTheRa, 2015-07-06:
I'm hoping one day someone will document. i guess most people just get in and play to figure it out

u8mydog, 2015-07-06:
There are files missing from the src in the repo (such as application.windows64/lib/MW_OSD_GUI.jar. These files do exist in the compiled download. When all the files are available, here's the contents of my .bat file to start the gui on windows7:

java -Djava.ext.dirs=lib -Djava.library.path=lib MW_OSD_GUI

ShikOfTheRa, 2015-07-06:
I'm not an expert on processing / java - but I don't think these files should be there in the source. ?? I would expect the compiler to create.

I'm guessing it is a java path issue as described in the FAQ. Maybe? Its a real issue with a number of applications. i see this in our work location too

closed 2015-07-20 by ShikOfTheRa