ShikOfTheRa/scarab-osd #85: can't set roll rate

dustin, 2015-07-04:
I'm running baseflight_pr and tried to adjust roll and pitch rate in the field.  Pitch rate took, but roll rate reset itself after change.

ShikOfTheRa, 2015-07-06:
OK - will take a look. Unable to test this at moment.

fly4props, 2015-07-08:
Hi ShikOfTheRa, I to am trying to set my yaw rate , roll/pitch rate via osd and it is just resetting back to what I have set it in the FC gui . Running CleanFlight 1.9 SPracingF3 . And thanks for all the hard work you put into this project!

dustin, 2015-07-08:
@fly4props, this sounds unrelated.  Different FC firmware, different
feature.

fly4props, 2015-07-08:
@dusten ahh I had mwosd 1.3 working so I could adjust pids. I thought this wold also work. no such luck, Ill just go back to the 1.3. Thanks

ShikOfTheRa, 2015-07-09:
@dustin - BF_PR now removed. I had a look at the code and they no longer change P+R using the same serial MSP. So only versions up to up to and including Stable 2015.05.13 are supported. Versions / release after that will not be supported. Sorry...

ShikOfTheRa, 2015-07-12:
@dustin 
Actually looking at the code, it appears 2015.03.27 is last full supported release. After that Pitch + Roll rates not supported... 

@fly4props latest repo should be working now

dustin, 2015-07-13:
Does this mean only separate roll and pitch rates are supported moving forward?

ShikOfTheRa, 2015-07-13:
At the moment it means unable to modify pitch and roll for releases after 2015.03.27
They have removed the ability to set teh P+R within the normal serial MSP where this is done.

I saw changes and added in the P+R code, but the baseflight guys subsequently removed it and put it in their own config serial MSP code. Its a pain in the ass for me, but understand why they did it. Its a different approach to Cleanflight. I am working on a solution

ShikOfTheRa, 2015-07-13:
Beta 4 here contains hopefully a solution for latest Baseflight release with separate RC pitch/roll rates.
I have tested tonight with a copter on the bench and it seems to work - reads the settings OK, made changes, wrote back and verified OK.

Test very carefully - there is always potential for bad results when saving - hold copter down to be sure! or disable props
https://github.com/ShikOfTheRa/scarab-osd/releases

ShikOfTheRa, 2015-07-20:
OK - had some confirmations its resolve din latest. I am using and have tested myself so pretty confident its OK

closed 2015-07-20 by ShikOfTheRa