ShikOfTheRa/scarab-osd #86: OSD stops displaying -- requires reboot

dustin, 2015-07-06:
I'm not entirely sure what leads to this, but I'll often have my OSD go out while the rest of my system continues to operate OK.  I've captured one case in particular today:

http://imgur.com/a/noDZ8

It looks like it may have browned out or similar (though I'm not entirely sure why).  Ideally, in this situation, one of two things could happen:

1. The OSD could detect the condition and reinit.
2. The OSD could allow me to tell it that painting is failing and reinit.

I suppose the second one can be accomplished by tickling the reset on the AVR.  But it's not clear to me exactly what's putting me in this state.  e.g. is it the MAX itself that's needing to be reinitialized (and can you tell?) or is it the OSD itself that could be fixed up with a watchdog?

ShikOfTheRa, 2015-07-06:
The max chip is really sensitive to voltage. Its especially tough if using servos.
I think it starts to fail around 4.85 v

Anyway I added some functionality in last week that might help to resolve. Its untested in real world - only tested by me on the bench. Its available on the repositor if you want to download the latest zip.Comablitble with 1.4.x GUI so dont need to recompile a new one: 
#define MAXSTALLDETECT              // Enable to attempt to detect MAX chip stall from bad power. Attempts to restart.

Love to heard feedback if it works. Would expect to see an outage of about 1 second when it drops / restarts.

dustin, 2015-07-06:
This looks great.  Not entirely sure what's causing the voltage to dip (or even if that's what's going on).  I added a 100µF cap to try to stabilize.  The last outage I had fell out with no visible artifacts on the screen.  One frame OSD, next NOSD.

Mine seems to be a little more stabilized, but I can usually get a failure every couple batteries, so I'm pretty excited to see this fix.

ShikOfTheRa, 2015-07-06:
Actually i run a 3000uh and an inductor on my test tricopter - only way to stop the servo noise.

i also have a huge  tricopter with 4okg servos on the boom - I have to run the OSD/CPU on a separate supply. Only way I can keep it working when yaw servo moves.

dustin, 2015-07-07:
I had to drop down to about 4.3V for this to fail, but on the previous version, it wouldn't come back and now it just does the right thing.  Very promising.

dustin, 2015-07-07:
OK.  I can confirm that this does indeed recover which is such a tremendous improvement.

However, the OSD comes back a bit off:

Before:

[image]

Loss:

[image]

After:

[image]

ShikOfTheRa, 2015-07-07:
Ok - thats great on the OSD restart. I tested also last night - quite pleased actually with the results.
This is what I use. My tricopter OSD resets every time I move teh tail servo. This capacitor stops completely. And is very light/ plugs in easy  - :
http://rctimer.com/product-1005.html

I didn't notice the screen offset like you have. I do not have  a clue of the cause / fix. Will test again at home to see if can replicate / fix

dustin, 2015-07-07:
The thing I'm doing this on is pretty small.  I'm using the witespy
microminimosd.  I've got a pololu 600mA 5V stepdown coming from 3S that's
also powering a CMOS camera, afro mini and an opensky rx.  I should
probably check the current a bit, but it's basically going now.

I put a 100µF cap directly on the OSD to help with blips, but I've not
quite figured out where the dirtiness is coming from.  There are a few
questionable components in there.  :)

Having the OSD come back, even if offset is a tremendous improvement.  I
only had the one failure, so I'm not sure if this is a 100%-of-the-time
thing, but it's at least possible.

Perhaps an explicit "please reset the OSD" signal might still be good if
they are going to be wonky on occasion.  That could, perhaps, be my third
HUD.

Thanks a ton for this work, though.

ShikOfTheRa, 2015-07-07:
I wonder if maybe the max chip has spurious data after a power out.
2 registers have vertical and horizontal offsets. They are not changed by the OSD, but possibly could be corrupted.

I have added a chip hard reset to simulate a clean power up. Maybe it will help. This is now on the repository - although not via the releases

ShikOfTheRa, 2015-07-10:
The hard reset is available in 1.4.2 beta 2 download here...
If able to test and confirm, that would be great
https://github.com/ShikOfTheRa/scarab-osd/releases

dustin, 2015-07-10:
I haven't been able to fly this new code yet, but I seem to have no issues
replicating the sync issue, so I suspect my next flight will confirm the
fix.

dustin, 2015-07-11:
I tried the beta. I rode pretty hard and only had one confirmed restart.
It did not go out of sync.

fly4props, 2015-07-19:
and here I thought I had a ground loop somewhere in my system!!! I too have been experiencing the loss of the  osd . I did remedy this my taking the ground from the vtx off, but I experienced to much noise. Ill have to try out the new version. Thanks ShikOfTheRa , not to fix all the wires and make things neat again.

ShikOfTheRa, 2015-07-20:
OK thanks also for feedback everyone. Seem slike we have a resolution.

closed 2015-07-20 by ShikOfTheRa