ShikOfTheRa/scarab-osd #95: Question: After flight statistics are cumulative or per flight?

truhlikfredy, 2015-07-20:
When you disarm your multi-rotor a flight characteristics page is displayed, fly time and some information on it. 

These information are stats on given flight? Fly time is reset every time, but for example **MAX AMPS** is not reset so it will display maximal current from the moment battery was connected and not from last flight. **MAH USED** and others are the same i think. 

Is this by design?

ShikOfTheRa, 2015-07-20:
All statistics are reset when battery changed by default.Nothing is stored for future flights

If "reset statistics upon arm" is enabled in gui, the following are also reset each time amed:
    trip=0;
    distanceMAX=0;
    altitudeMAX=0;
    speedMAX=0;
    ampMAX=0;
    flyingTime=0;

No

truhlikfredy, 2015-07-20:
Oh, ok. Thanks. I didn't noticed the reset statistics on the first glance.

closed 2015-07-20 by truhlikfredy