ShikOfTheRa/scarab-osd #97: GUI Freezes

u8mydog, 2015-07-25:
I just downloaded MASTER.

As soon as the GUI connects, it freezes. I can't enable/disable options or click on load or save buttons.

The only config.h changes I've made are to enable the following:
WITESPYMICRO (and disabled MINIMOSD)
CLEANFLIGHT190 (and disabled MULTIWII)
INTRO_MENU (and disabled INTRO_VERSION)

When the GUI starts, it does connect to the board and load the configurations from the board. I just can't change anything. Also, both leds flash rapidly on the ftdi board so I'm guessing that the MSP info is flowing.

I've tried both application.windows32 and 64bit on a windows7 machine. I then tried the GUI for the K.... R1.3 and I was able to read the config from the board and adjust settings but the gui simulate screen said 'disconnected'.

Any ideas?

I'm going to revert config.h to see if that makes a difference.

u8mydog, 2015-07-25:
Tried reverting to original files and gui still freezes after loading settings from the board. ftdi tx led flickers such that led is dim and rx led is solid. I then tried gui from build 1.4.1 and the GUI appears to work and the ftdi tx led is dim and rx led is not on.

readerror67, 2015-07-26:
Can confirm, 1.4.2.8 I am unable to click any options after it loads. Using 1.4.2.8 firmware on R1.4.1 works fine though.

ShikOfTheRa, 2015-07-27:
Not really sure why - i am not seeing this however with some of the recent changes it would be good to try this version (update the OSD and GUI):

https://github.com/ShikOfTheRa/scarab-osd/releases/tag/v1.4.10

I have tried this with the following to be same... 
WITESPYMICRO (and disabled MINIMOSD)
CLEANFLIGHT190 (and disabled MULTIWII)
INTRO_MENU (and disabled INTRO_VERSION)

u8mydog, 2015-07-28:
The v1.4.10 build works. I do notice two differences between master and 1.4.10 application dir. 

master has file application.windows32\data\gui.cfg but v1.4.10 doesn't.

And application.windows64\data\gui.cfg has some differences.

As soon as I disabled AutoSimulator in MASTER application.windows64\data\gui.cfg, then the MASTER gui app started to work.

u8mydog, 2015-07-28:
Opps. You can see the GUI freeze with v1.4.10 by connecting to the com port, then enable 'simulate on OSD'.

ShikOfTheRa, 2015-07-28:
OK this is positive.
1 at least can use the GUI for setting up.
2 narrows down area to look!

ShikOfTheRa, 2015-07-28:
When it locks up...
In the GUI under the "LAYOUT" box do you see MSP: with some numbers.
Are they changing constantly before it locks up. And what happens after - does it still lock? IS it the same one always displayed when locks up?

u8mydog, 2015-07-29:
There is a MSP: value that flickers from 100 to maybe 120. As soon as I press 'simulate on osd', the MSP value freezes.

u8mydog, 2015-07-29:
I just saw your 1.4.2.11 and tried it. There is no change. It still reacts as previously mentioned.

u8mydog, 2015-07-29:
Sorry, one more thing. Just to be clear, I only have the FTDI connected to the micro minimosd. There is no video src connected.

ShikOfTheRa, 2015-07-29:
in gui.cfg in data folder of latest version, set sim type to 0. That will probably fix it. 

OK - could you tell me a bit about your pc - 
language
what locale settings are - french / us etc. etc.
java  version and if 32 or 64 bit

I want to understand why it will not run when it does on most pc's

u8mydog, 2015-07-29:
symType=0 works but the MSP value updates much slower.
English
US
C:\Windows\System32>java -version
java version "1.8.0_40"
Java(TM) SE Runtime Environment (build 1.8.0_40-b26)
Java HotSpot(TM) 64-Bit Server VM (build 25.40-b25, mixed mode)

u8mydog, 2015-07-29:
Just tried following with no change:
java version "1.7.0_75"
Java(TM) SE Runtime Environment (build 1.7.0_75-b13)
Java HotSpot(TM) 64-Bit Server VM (build 24.75-b04, mixed mode)

ShikOfTheRa, 2015-08-10:
I dont know why this is affecting you - and one other, but will leave symtype=0 as default for now

u8mydog, 2015-08-12:
Can you tell me what differences occur with symtype=1? For example, is tcp/ip used to communicate with localhost?
My machine is an engineering workstation laptop and is managed by a remote I.T. group. So I have corporate grade software firewall and virus scanner installed. Sometimes the firewall can get in the way.
The machine also has multiple versions of tcl, java, etc installed. So for example, if symtype=1 causes a search for java.exe instead of using the windows path, it could pick up a different version of java.exe. I did a search for java.exe and stopped counting when I hit 24 java.exe files found.

ShikOfTheRa, 2015-08-12:
simtype=0 continually sends all data irrespective of whether the OSD is requesting it or not

simtype=1 only sends the data requested by the OSD and only sends it once per request. This replicates real life much more closesly. The actual underlying code is shared between both. Its why so difficult to determine the issue.

I searched my pc - I only have 1 java.exe
I think maybe that is indicator of an issue? Thats never correct. Maybe there is some mixup.

If it were me I would note what is installed then uninstall everything and install either a single 32 bit or a 64 bit version
1 - Use this to remove.
2 - Edit the system variable "path" and remove all references to java
3 - install latest java.

ShikOfTheRa, 2015-08-12:
oops - this is what mean to say to run
https://singularlabs.com/software/javara/

u8mydog, 2015-08-14:
Unfortunately, I can't uninstall all the jre's on my machine without having an ill effect on many different versions of engineering tools on my machine. Each version of eng tool has its own local install of java. Some of these eng tool suites include different versions of eclipse and custom plug-ins.

u8mydog, 2015-08-14:
BTW: javara shows that I have Java 6 update75, java 7 update 75 and java 8 update 40 installed.
My environment points to java 8. I will try updating to java 8 update 41 but will keep the other versions installed on my machine.

   

https://singularlabs.com/software/javara/—

ShikOfTheRa, 2015-08-14:
I was nervous to uninstall all my various java versions ( I had  a few too although not 25!!) Removed all then put latest on. Have had no issues. And use quiet a lot that are dependant upon java

closed 2015-09-30 by ShikOfTheRa