ShikOfTheRa/scarab-osd #99: Current readings

readerror67, 2015-07-26:
Maybe I'm too tired to think properly, but the latest beta the current readings seem to be very off.

The change seems to divide the 10mA reading by 10 when it seems the base value is 100mA, shouldn't it multiply?

https://github.com/ShikOfTheRa/scarab-osd/blob/1d52c49137fc4c9943d88524650a4146304b6ab1/MW_OSD/MW_OSD.ino#L847-L848

ShikOfTheRa, 2015-07-27:
I think by default:
multiwii (pre 2.4) sends in 1ma increments. Therefore if current is 2.5A, FC sends 2500
Baseflight/Cleanflight send is 10ma increments. . Therefore if current is 2.5A, FC sends 250
multiwii (2.4) sends in 100ma increments. . Therefore if current is 2.5A, FC sends 25

So in case of Baseflight, it sets amperage in OSD to 250 for 2.5A
Now if baseflight is defined,, so is amperage_10ma. This should then divide the 250FC amperage by 10 to get 25

Now when it displays, it sets the least significant digit to be 1st decimal place. So 25 becomes 2.5A on display

I have never used a current sensor so I can’t say I am sure this is correct! Makes testing a little challenging too.
And also I think other issue is in BF/CF there is an option to change and send different value instead of its default of 10ma.

If understanding is incorrect, feel free to let me know!

ShikOfTheRa, 2015-08-10:
Have you been able to test and verify allis correct with the latest?

closed 2015-09-30 by ShikOfTheRa