ShikOfTheRa/scarab-osd #100: Java 8 causes exception during GUI startup

JoeB99, 2015-07-31:
Win7, x64

When starting the GUI, a java 'animation' exception is thrown in the console window.

Uninstalling Java 8, and installing Java 7 update 80 fixes the issue.

ShikOfTheRa, 2015-08-03:
I am using 1.8 !
The messages above are also what you would see if running a 32bit gui on 64 bit java or vice versa.

JoeB99, 2015-08-03:
1.8 ? The latest version in the downloads section is R1.4.1.
Is there a new release?

ShikOfTheRa, 2015-08-03:
Sorry for confusion - Java 1.8 
MWOSD R1.5 will be out soon. Just waiting for completion of testing for various platforms.
Will be very close to this beta test release...
https://github.com/ShikOfTheRa/scarab-osd/releases/tag/v1.4.2.11

closed 2015-08-10 by ShikOfTheRa