ShikOfTheRa/scarab-osd #102: Feature: List of elements in the layout editor

JoeB99, 2015-08-04:
Instead of a + and - button to change between elements that can be turned on/off in the layout editor, can there just be a list so all element states can be seen, with a tick box next to eack one to turn them on and off. This will make selecting an element, moving it and turning them on and off much much! quicker when setting up custom layouts.

ShikOfTheRa, 2015-08-04:
Agree its crap. Its a quick and dirty solution purely to bring us functionality we didn't have before.

I was hoping that someone will write a new app with this type of functionality included. Also drag & drop on screen.

So right now I'm waiting for someone to do this as unfortunately I don't have the time.

JoeB99, 2015-08-04:
Maybe this will be of use?
https://code.google.com/p/minimosd-extra/
They have an editor that can do all this.

/ JoeB

Date: Tue, 4 Aug 2015 03:27:47 -0700

ShikOfTheRa, 2015-08-04:
No use at all - its completely different. 
But its the sort of functionality that would be great to have for MWOSD

ShikOfTheRa, 2015-08-10:
Have added this to the wish list. I hope one day someone will pick this up

closed 2015-08-10 by ShikOfTheRa