ShikOfTheRa/scarab-osd #106: PIDs in OSD menu to floats

NL-Splinter, 2015-08-13:
Coming from BorisB's work on PID controller improvements came the need to show PIDS in floats in the CF configurator. Though this is not yet in master, it's likely coming. It would be great if MWOSD will also be able to show the PIDs as float (or add just 1 extra digit, ie. P=1,02 -> MWOSD: 102)

CrazyCoder, 2015-08-13:
+1, see https://github.com/borisbstyle/cleanflight-configurator/commit/11b9e6f5cad72e041025b2d3a748e782aaec91b3

Also see how OSD is done in Vortex, it can tune floats no problem.

ShikOfTheRa, 2015-08-13:
Had a quick look. Not sure what some of the stuff is as I don't have time to follow cleanflight (wish I did....)
Hopefully someone will provide the code and create a pull request for this

CrazyCoder, 2015-09-03:
Cleanflight will probably move all PID controllers configuration to floats at some point, not just LuxFloat. This will unify the Configurator and make it more easier to support in OSD code.

ShikOfTheRa, 2015-09-04:
I would say easier !!!
 -when all the others FC's dont use floats 
-atmega chip doesn't have much resource left:(

But mucho easier if CF is all one way or the other.

ShikOfTheRa, 2015-10-21:
Added this to the wish list / roadmap.
Note I don't have any plans currently to do this due to lack of time.

Removing this from issues list as in the roadmap

closed 2015-10-21 by ShikOfTheRa