ShikOfTheRa/scarab-osd #109: Forced crosshair depends on accelerometer

CrazyCoder, 2015-08-23:
At the moment if you disable accelerometer (needed to get 1000 loop time on F1 hardware for gyro synced looptime feature), crosshair enabled by `FORCECROSSHAIR` define doesn't work as it's called inside `displayHorizon` function which is not used without accelerometer.

I suggest to make a separate function `displayForcedCrosshair()` like this:

```
    uint16_t position = getPosition(horizonPosition)-(2*LINE);
    screen[position+2*LINE+7-1] = SYM_AH_CENTER_LINE;
    screen[position+2*LINE+7+1] = SYM_AH_CENTER_LINE_RIGHT;
    screen[position+2*LINE+7] =   SYM_AH_CENTER;
```

and call it from `MW_OSD.ino` surrounded by `#ifdef FORCECROSSHAIR`. Just after `displayHorizon` outside of `if(MwSensorPresent&ACCELEROMETER)` block.

This way crosshair will not depend on what sensors you have enabled. I've already tried it and it works just fine. Should be an easy and safe fix.

ShikOfTheRa, 2015-08-24:
yes looks an easy no issue change. Will look at in next couple of days

ShikOfTheRa, 2015-10-21:
OK - finally added !!!

closed 2015-10-21 by ShikOfTheRa