ShikOfTheRa/scarab-osd #110: Move crosshair with camera gimbal

smitty97, 2015-08-24:
I have a pitch gimbal on my camera that i can move either manually or with cleanflight's camstab feature.  Can the crosshair (and AHI) move up and down on the screen to maintain some truth to which way the craft is pointed at?  I'd be happy even if it had to move by a full text line at a time.

ShikOfTheRa, 2015-08-24:
This is possible, but with a few negatives:
1 - one character row at a time resolution at a time. I think +/- 4 characters from midpoint which is not great so maybe every 10 degrees
2 - needs to be able to read an rc channel which is used to control the gimabl. i.e. this must also be connected and visible to the FC

I can add to the wishlist, but be prepared for a very long wait if it ever gets done at all. So many requests...

smitty97, 2015-08-24:
Yep, that sounds perfect.  I almost never fly with the fully active gimbal, so moving the servo to a set position and having the center hop down a text row at a time is fine.  The naze32 fc controls the servo so the data should be in there somewhere to read.  Thanks!

ShikOfTheRa, 2015-08-27:
Thinking about this - the whole row will move up or down. It will significantly reduce the amount of area the horizon has to operate.

It OK to do this way, but can also....
Move just the center crosshair.
Put in  a separate indicator for pitch. Some other OSD's have this. Usually a line up the side of display.

Must admit I am not sure which is best as none of them really do it for me. I prefer the cross hair and ahi staying in the center :)

smitty97, 2015-08-27:
I think the whole row should move up or down.  On a quadcopter with the camera pitched up 30deg for example:

At rest on the ground, the center and AHI will be 3 rows down from normal.  The AHI has limited room here, but that's fine.

In flight, since the craft has to be pitched down to move forward, the center remains 3 rows down but the AHI has automatically moved back up to screen center, following the horizon.  If i do a snap roll, the video should appear to rotate around the center mark.

ShikOfTheRa, 2015-10-21:
Added this to the wish list on development roadmap. Removing form issues list.

closed 2015-10-21 by ShikOfTheRa