ShikOfTheRa/scarab-osd #111: Mac GUI fails to launch

cam-smith, 2015-08-26:
attempting to run the latest version, the mac gui fails to launch and prompts the user to delete the app.

The previous version listed on the site worked

ShikOfTheRa, 2015-08-27:
Hello - an updated MAC GUI was committed by someone else as I do not use MAC. I have compiled a new one and uploaded as have done in previous way. Please can you test and advise. Its in here...:
https://github.com/ShikOfTheRa/scarab-osd/archive/master.zip

cam-smith, 2015-08-27:
I've tried running this one, and it doesn't work either... when I run it with the console active I get the following error message:

27/08/2015 10:34:53.350 pm CoreServicesUIAgent[18569]: Error -60005 creating authorization
27/08/2015 10:34:53.363 pm CoreServicesUIAgent[18569]: Error SecAssessmentCreate: The operation couldn’t be completed. (OSStatus error -67054.)
27/08/2015 10:34:54.924 pm com.apple.xpc.launchd[1]: (com.apple.xpc.launchd.oneshot.0x10000030.MW_OSD_GUI[18584]) Service exited due to signal: Killed: 9

ShikOfTheRa, 2015-08-28:
Unfortunately I can't help with this. I dont have a real mac!. I do have OSX Mavericks running in a virtual machine on my PC. It runs OK on that.

Error message looks like its security related.
maybe something in FAQ 4+5

ohhorob, 2015-08-29:
I'll take a look...

ohhorob, 2015-08-29:
@cam-smith could you give a bit more info.. did you `git clone`, or zip download to get the GUI app on your Mac? Which branch/tag/revision was that from??

ohhorob, 2015-08-29:
I just verified that it's working for me on Mac OS X 10.10.5, using a git checkout of master branch.
At any rate, open the terminal and check the executable bits are properly set:

    scarab-osd rob$ ls -alh MW_OSD_GUI/application.macosx/MW_OSD_GUI.app/Contents/MacOS/
    total 128
    drwxr-xr-x  3 rob  staff   102B Aug 29 09:55 .
    drwxr-xr-x  8 rob  staff   272B Aug  8 10:47 ..
    -rwxr-xr-x  1 rob  staff    63K Aug 29 09:55 MW_OSD_GUI

Make sure there are "x" letters in the last line.

I don't get any errors trying to open it

    scarab-osd rob$ open MW_OSD_GUI/application.macosx/MW_OSD_GUI.app

joshtbuff, 2015-08-31:
I cannot get this to launch either.

Failed to launch user should move to trash. I downloaded the zip. scarab-osd-R1.5

ShikOfTheRa, 2015-09-01:
Do you have java installed? Its a pre-requisite. java.com for downloads.

joshtbuff, 2015-09-01:
The whole Java suit is installed.

ShikOfTheRa, 2015-09-01:
mmm I can only suggest googling that... I know nothing about mac's

anything in here?:
https://www.java.com/en/download/faq/java_mac.xml

nathantsoi, 2015-09-01:
fwiw, the mac app works fine for me.

joshtbuff, 2015-09-01:
Did you do anything other than download and then double click?

joshtbuff, 2015-09-01:
Mine is working now. I uninstalled and reinstalled Java.

cam-smith, 2015-09-01:
@ohhorob I've downloaded the package on the public page, I've downloaded the zip linked to by @ShikOfTheRa neither of them worked.  I've checked the execution bit, and it matches the example given, I've attempted to run it via sudo. I've uninstalled java, then reinstalled java... it seems that it just wont work for me.  Other java apps work on my machine, and the 1.4.1 version linked in the Downloads.md file from the webpage worked

joshtbuff, 2015-09-02:
What is your error. I've just worked through this and I am willing to offer some help although it isn't much.

nathantsoi, 2015-09-02:
@cam-smith, i cloned the repo and ran the app straight from the source:

```
git clone https://github.com/ShikOfTheRa/scarab-osd.git
open scarab-osd/MW_OSD_GUI/application.macosx/MW_OSD_GUI.app
```

my java version is 1.8

```
› java -version
java version "1.8.0_45"
Java(TM) SE Runtime Environment (build 1.8.0_45-b14)
Java HotSpot(TM) 64-Bit Server VM (build 25.45-b02, mixed mode)
```

ShikOfTheRa, 2015-09-02:
@cam-smith as hinted above maybe if cant resolve then one approach is to install "processing" and run the gui source code directly. Its very easy. Just like arduino... open up the source code (in other/mwosd_GUI and click the play button

ShikOfTheRa, 2015-09-02:
@joshtbuff thanks for feedback so we know for next poor mac user! Always wanted a mac. but no more...

closed 2015-09-30 by ShikOfTheRa

keithngkw, 2016-10-06:
I solved the problem by downloading the latest version of Java

yxeGithub, 2017-01-13:
I have this working by just changing the execute permissions.

I ran ls -alh MW_OSD_GUI/application.macosx/MW_OSD_GUI.app/Contents/MacOS/ and noticed it didnt have the corect permissions so I ran: chmod 755 MW_OSD_GUI/application.macosx/MW_OSD_GUI.app/Contents/MacOS/MW_OSD_GUI and then I could open it from finder or the command line.

Just wanted to update this thread for the next time I come across it.

ShikOfTheRa, 2017-01-13:
Thanks - added to FAQ. Chrome app will be released soon. 

4 I can't open GUI on my MAC - unidentified developer error / unable to open

http://www.imore.com/how-open-apps-unidentified-developer-os-x-mountain-lion
Ensure the application.macosx/MW_OSD_GUI.app/Contents/MacOS/MW_OSD_GUI file is executable
If not issue chmod +x on that file
chmod 755 MW_OSD_GUI/application.macosx/MW_OSD_GUI.app/Contents/MacOS/MW_OSD_GUI