ShikOfTheRa/scarab-osd #112: Power/speed value for optimize criuse speed in airplane.

sppnk, 2015-08-27:
Hi Shikra,

To get good range with airplanes if could be very interesting to see on screen the Watts/speed value, so if the pilot or future autopilot modes (cruise speed) can get the lower value here, they are getting the optimal speed for long range porpouses.

 I know it can be made by debug[x] values, but perhaps it could be included in the main code. Should be something like (voltage * amperage) / speed, and a simple filtering of that. Thx.

ShikOfTheRa, 2015-08-27:
Hello - yes this is in the roadmap / might even make it to next release.

https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/ROADMAP%20%26%20BUG%20TRACKER.md

sppnk, 2015-09-24:
@ShikOfTheRa, please note that I mean the value of the division of power by speed (not power or speed). The minimum of that operation should be teoretically the optimal cruise speed for longer flights. Thx.

ShikOfTheRa, 2015-10-21:
Am closing this as its on the wishlist/roadmap. Noted reference to watts/hr/km as options

sppnk, 2016-08-22:
Hi Shikra, is this feature of Wh/km already in the code? I is very interesting for fpv long range fyiers. Thks.

ShikOfTheRa, 2016-08-25:
If feeling brave, can try this.. I haven't tested myself! Next week before 
//#define DISPLAYEFFICIENCY will give watts/speed if enabled. Also need to turn on the power element in the layouts so it will be displayed

sppnk, 2016-08-26:
Great !

ShikOfTheRa, 2016-09-07:
OK. Revised and tested by schalonsus. This can be closed :)

closed 2016-09-07 by ShikOfTheRa

sppnk, 2016-09-12:
Hi, I have tested this, but I only get values like 0 and 1, and nothing else. When the plane is stopped I have 999, what is ok because speed is zero, but whith 50km/h and 7 amps on 3S batt, I should have 1.56 Wh/km aprox.

So I guess the only problem is that there are not decimals in the value showed.

...and I dont see armed/unarmed status now. May be the watt/speed value is overlapping it?

ShikOfTheRa, 2016-09-12:
Can you try the latest version from repository - sound like you have the first test version. The latest one verified by schalonsus will show it formatted - e.g. 1.56 instead of 156. It will need to be enabled in the config.h and layouts (uses Watts field) . It is hidden when not armed or at 0 speed.
I just tried on simulator and seems to work with your values. It shouldn't overwrite the warning status area. Suggest to check that is also enabled in the layout editor and write the layouts to OSD again to be sure..

sppnk, 2016-09-12:
Thanks, I will do, but I downloaded the repository after read yor message about the tests from schanlonsus. may be it is a newer version. 

In config.h I just defined DISPLAYEFFICIENCY (no DISPLAYWATTS), then in configurator I enabled in layouts. Will try again.

ShikOfTheRa, 2016-09-12:
If its displaying without a "." it must be the earlier version. Let us know how it goes. Fingers crossed....

sppnk, 2016-09-13:
Here is what I see, I see a star '*':

[image]

ShikOfTheRa, 2016-09-13:
This is what should see... R 1.6.2.0 on startup
Other thing - possibly something overwriting?

Possible its older version ( I see 1.6 for at 36kmh for those values)
[image]

sppnk, 2016-09-13:
I will check the version number on startup and report.

sppnk, 2016-09-13:
Hi. The version is R 1.6.2.0 on startup. 

I guess I should try to move the value to other place of the screen or something.

sppnk, 2017-04-10:
Hi, this is old, but just to report here that I have actualized recently and I can see the Wh/km indicator working fine !! My Mini Talon climbs at high values, but cruises at 1.0 Wh/km or less if has wind in tail. 
Very useful to optimize long range flights. Great !!

ShikOfTheRa, 2017-04-10:
Brill. i often wonder how often some of these functions are used...

sppnk, 2017-04-10:
Yes. I guess the long range pilots who can report about this things mainly are using other gear than this (vector osds and so on). 

But with iNAV this is changing, so I hope more people enjoying this kind of features in their minmosd's..

ShikOfTheRa, 2017-04-10:
Yeah - huge change recently. iNav is impressive. A lot to learn though...