ShikOfTheRa/scarab-osd #114: Mission mode activated when using telemetry switch on Cleanflight

Netsplits, 2015-08-30:
I've setup on Cleanflight 1.9 a switch to enable telemetry, even in disarmed mode, and when I set the switch on, the OSD unexpectedly enters MISSION mode. I've something wrong on my configuration?

ShikOfTheRa, 2015-09-01:
Just looked- it appears cleanflight are using same status bit for telemetry as multiwii uses for GPS waypoints. I'll have to make amendment to accommodate this.

ShikOfTheRa, 2015-09-01:
Have made some changes - please test to see if it works.
GUI is same - just need to upload new sketch to OSD
 
https://github.com/ShikOfTheRa/scarab-osd/archive/master.zip

closed 2015-09-30 by ShikOfTheRa

Netsplits, 2015-10-10:
Sorry for the late feedback, but it works perfectly now. Thanks.

Netsplits, 2015-10-12:
Don't know if it's because I've just upgraded to Cleanflight 1.10 version, but mission appears again, not sure what did activate it, I suppose IDs have changed again on Cleanflight code...