ShikOfTheRa/scarab-osd #115: Osd not start

lucafedechen, 2015-09-07:
Hi..
I have a error..I use microosd and flash latest release, I use átom cc3d with boris b. beta cleanflight,
Camera ntsc, microosd connect to main port..ground,TX and rx..VLC i connect from 5v polulu step down..polulu power flight control, rx and osd only..camera it's power from another 5v lineal regulator.
Osd start and display data, when I disconnect my rx radio from cc3d, when connect the rx the osd not start and the led in microosd only blink 1 time, disconnect the main port the osd start and shows no data available..
Check all voltages and not find issue..the osd start and show data only when rx disconnected
Only for test I flash rushosd and works ok..no problem at all.
I hope for you's help..any idea to solve?? 
Have nice day
Regrds

ShikOfTheRa, 2015-09-17:
Have you checked FAQ?

Most likely - blank screen layout selected

ShikOfTheRa, 2015-09-17:
Just looked at code - actually everything is managed by "ontime" setting.

closed 2015-09-30 by ShikOfTheRa