ShikOfTheRa/scarab-osd #117: Docs for success with d4r RSSI?

dustin, 2015-09-13:
I've got a d4r (high speed pwm) hooked up to an OSDoge.  I've tried several combinations of things to get the RSSI reading and the most success I've had gives me very coarse RSSI readings (essentially 100% or < 100% if things start going bad).

OSDoge has an RSSI filter option, but enabling that prevented me from ever seeing anything useful.  It's unclear how to enable analog RSSI with the filter.

If it works, a FAQ entry for how would be good.

ShikOfTheRa, 2015-09-14:
The Frsky stuff doesn't do normal PWM RSSI. I have contacted them but they will not reply.

I think the only answer is to follwo case 2 here....
https://code.google.com/p/minoposd/wiki/AnalogRSSI

ShikOfTheRa, 2015-10-19:
Updated documentation in the FAQ for FRSKY RSSI options
Personally , I think the best approach is to convert to analogue with a filter...

https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/FAQ.md

closed 2015-10-19 by ShikOfTheRa

dustin, 2015-10-19:
Thanks a lot for this.  I think the telemetry based ones are particularly bad.  Filter stuff seems like the best bet overall.