ShikOfTheRa/scarab-osd #118: Cannot make "Fly Time" visible in Layout Editor

joshuabardwell, 2015-09-17:
When I enable "Fly Time" in the layout editor, the field does not appear on the layout. Other values seem to depend on switches as to whether they appear or not, but I can't find a switch that corresponds to Fly Time.

ShikOfTheRa, 2015-09-17:
On GUI it is labelled "Display Timer"
In Layout editor there are two - "FlyTime" and "Ontime". Its from legacy code which need to be consolidated to a single position at some point.
Both need to be enabled and likely in same position. 

In code, both clocks are running, but only one is ever displayed. When armed the flytime is displayed. Else its on time

joshuabardwell, 2015-09-17:
>When armed the flytime is displayed. Else its on time

Ahh. Thanks. That was the piece I was missing.

closed 2015-09-17 by joshuabardwell