ShikOfTheRa/scarab-osd #119: Artificial horizon lags behind actual attitude

DTFUHF, 2015-09-21:
As copters get faster and faster the AHI lag becomes more noticeable. Requests have been made to speed up this visual aid.

I've noticed that the AHI is updated in the same loop with RC inputs, PIDs, and other items the user probably wouldn't notice being updated a bit slower than they are now. If these items were given lower priority, I bet the attitude could be updated at a rate close to VSYNC.

Chrischhan, 2015-09-21:
i searched also for an solution for slow horizon update, my first look was in the config file for some parameters i didn't know. 
I found the MSP Speed settings and looked further.
(did i found a typo on: https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Def.h#L230 MSP_SPEED_LOW rather ...MED?)
is this an option to speed up the Horizon?

ShikOfTheRa, 2015-09-29:
Good catch on typo - thanks chris - will address.

DTUHF - sure can do something here...
Currently updates 10 times per second. Lets provide option to do up to 25/30.

BUT - note that a lot of this delay is the process of requesting from FC and then waiting for a reply. If FC could have an option to keep sending at some hz factor, this would remove lag considerably. The OSD will process it regardless of not requesting...

Also, If using high frequency RSSI PWM this can slow it all down. If can run it at standard 50hz, that will improve. Am considering implementing something to mitigate this.

ShikOfTheRa, 2015-09-30:
Amendments made to try out.....
In practice I don't know how well the FC will be able to process  at higher rate. Might depend upon what else it is doing. It should drop some requests if the serial buffer fills up.

/******************** Serial MSP speed settings *********************/
// Choose ONLY ONE option: increases speeds of serial update - but with impact to flight controller 
//#define MSP_SPEED_LOW     // Enable for soft serial / slow baud rates.
#define MSP_SPEED_MED       // Default
//#define MSP_SPEED_HIGH    // Enable for faster AHI and speed updates. Requires higher baud rates and increases overhead on the FC to process

This provides guidance of expected update speed...
/********************  MSP enhancements rule definitions  *********************/
#if defined MSP_SPEED_HIGH
  #define hi_speed_cycle  10  // updates everything from serial approx 6.3 times per second, updates attitude 30 times per second
#elif defined MSP_SPEED_MED
  #define hi_speed_cycle  50  // same as low, but also updates attitude 10 times per second
#else
  #define hi_speed_cycle  50  // updates everything from serial approx 1.3 times per second
#endif

ShikOfTheRa, 2015-10-02:
I can't test on a controller for a couple of weeks, but seems to work in simulator as far as I can determine

ShikOfTheRa, 2015-10-05:
I also made amendments to RSSI code for those using high frequency PWM. It should speed up everything else

ShikOfTheRa, 2015-10-21:
Any chance for anyone to test?

DTFUHF, 2015-10-21:
This is working great for me, when MSP_SET_HIGH is enabled the AHI is AMAZINGLY fast.

Thanks for this.

ShikOfTheRa, 2015-10-21:
Awesome. It seemed to work for me so hoping to hear that from independant view!!

closed 2015-10-21 by ShikOfTheRa