ShikOfTheRa/scarab-osd #120: 20+ seconds are needed after power up before cleanflight 1.10 can be armed

vratinov, 2015-09-22:
I'm running pre-release 1.10 builds and I noticed that with 1.5 MWOSD connected it takes 20+ seconds after power up before the quad can be armed. Disconnecting the OSD allows me to ARM the quad instantly. This does not happen with 1.9.0.

ShikOfTheRa, 2015-09-29:
Not tried myself - or able to for a few weeks. Had a quick look at the commit log for CF. I wonder if it's something to do with this? Not quite sure why. Probably unrelated 

https://github.com/cleanflight/cleanflight/commit/3b9cd37a57ade925be22b322d96a8c36656b2f33

ShikOfTheRa, 2015-10-19:
IS this still an issue with anyone? Have had no other reports

jaffadog, 2015-10-21:
I'm running CF 1.10 and MWOSD master from a few days ago and have no such issue.

vratinov, 2015-10-21:
It appears to be resolved. I tested it the other day with latest branch and beta build of mwosd and didn't have an issue.

ShikOfTheRa, 2015-10-22:
Excellent - thanks

closed 2015-10-22 by ShikOfTheRa