ShikOfTheRa/scarab-osd #123: No overlay with camera in

presidenten, 2015-09-28:
Ive got a weird issue with my osd. I only get overlay when my camera is not connected. Then I can access naze settings and everything, but when the camera is connected I only get the video without overlay... :/

Anyone who have had the same issue or maybe know how to fix this?

The camera is a sony super had 650tvl which is set to lcd output.
I have tried mwosd fw 1.4.1 and 1.5.
I have tried both pal and ntsc settings on the osd on both firmwares.

ShikOfTheRa, 2015-09-29:
Normally that's wrong NTSC/PAL settings - so would suggest to double check. Change, save and then power cycle to be sure.

If that doesn't work, try disabling VSYNC in config.h

So far I haven't seen a camera that doesn't work - although some are incompatible and give a feint  flickering display.

presidenten, 2015-09-29:
Got a tip from a guy over at fpvlab, the solution was to solder the cameras gnd to the micro minim osds input gnd.
Thanks for the heads up anyways. We can close this issue.

closed 2015-09-29 by presidenten