ShikOfTheRa/scarab-osd #125: cleanflight "autotune" mode displayed as "land"

jaffadog, 2015-10-04:
I am using mwosd 1.5 with cleanflight 1.9 on a naze32.  When I put the naze into autotune mode, the osd displays the mode as "LAND".  Not a big deal... i don't plan on being in autotune much ;).

ShikOfTheRa, 2015-10-05:
I think it relates to this one. Can try by downloading latest master zip. Its in here if not sure where taht is
https://github.com/ShikOfTheRa/scarab-osd/issues/114

jaffadog, 2015-10-16:
I installed the latest master, and the OSD now displays "ACRO" when no mode is enabled (so its actually in acrobatic mode), and still displays "ACRO" when autotune is enabled.  Not a big deal.  Like I said, i expect autotune mode is seldom invoked - so probably not necessary to call out the mode in the OSD.

ShikOfTheRa, 2015-10-19:
Yes - that is the default - gyros always on - in whatever mode. I'd rather call it gyro, but most know it as acro so best leave it.

Only level below that is "passthrough" - no sensors enabled. This is only valid in baseflight /multiwii fixed wing though.
 
Like you say - its used for such a short time, not worth losing memory resource to implement.

At least LAND is gone!!

closed 2015-10-19 by ShikOfTheRa