ShikOfTheRa/scarab-osd #126: Voltage showing zero on cleanflight

bk79, 2015-10-11:
@ShikOfTheRa I've tried compiling the latest revion with cleanflight and battery level from FC shows zero on cleanflight 1.11 I've correctly configured the config.h to be working with cleanflight decommenting cleanflight only

ShikOfTheRa, 2015-10-11:
Is it only an issue with 1.11?

bk79, 2015-10-12:
No idea i'm using this release  now never tried pm previous one

ShikOfTheRa, 2015-10-12:
Have you selected the correct board? 

If still same, try connecting to bat2 - some boards have tave bat1/bat2 reversed

bk79, 2015-10-12:
Nope I'm using FCC values

ShikOfTheRa, 2015-10-12:
Ha - I didn't read your initial comment fully
Just tested on simulator and it work ok here. This means OK in Multiwii, baseflight and older cleanflight.
I don't know if 1.11 has changed. I'll be upgrading at some point but not sure when can test.

ShikOfTheRa, 2015-10-19:
Hi is this still an issue?

bk79, 2015-10-19:
I'll tell you in 20 days I burned my miniosd so waiting for a new one to
arrive from china

dustin, 2015-11-24:
@bk79 have you confirmed your flight controller was receiving the correct values itself?

closed 2015-11-30 by ShikOfTheRa