ShikOfTheRa/scarab-osd #129: TauLabs Support

dustin, 2015-10-19:
TauLabs flight control firmware is amazing and everybody should use it.  However, other than BrainFPV, there's not really a good OSD available.  I think MWOSD is an excellent option.

I've been developing a module for TL that allows MWOSD to talk to it as if it's talking to baseflight or similar.  My goal is to make it work with as few changes to MWOSD as possible.  I would, however, like it to be a supported target for MWOSD just to keep things clear.  e.g. `#define TAULABS` to pick out the bits and clarify how some MSP frames are handled (e.g. the kinds of differences you see between baseflight, cleanflight, and multwii).

I am actively working on this on the TL side, and if it's considered worthwhile as a target here, then I'll do the work for MWOSD to mix the best of both worlds.

ShikOfTheRa, 2015-10-20:
Hi – more than happy to have this included!!!  Must admit know very little about TL. Doesn’t seem to have a big following on the forums. Its pedigree looks good though.

Up to you how to implement – either to simulate multiwii MSP or put own  request / receive compatible with existing TL.

 If can keep the memory requirements as low as possible using defines it would be appreciated. Am trying to keep 1k free at compile of the default values to allow for those that wish to enable additional features. Also have to watch the stack and heap memory – its been reduced to zero several times with submissions hence the memory checking at the beginning.

This should be good!

This may help:
For the Naza,  basically everything is kept separate within its own  and then there is a table that refreshes the values similar to below.
Naza is currently not included due to lack of interest but I am maintaining it for now outside of the repository for personal reasons.

      GPS_numSat =
      if (GPS_numSat >= MINSATFIX){
        #ifdef GPSACTIVECHECK
          timer.GPS_active=GPSACTIVECHECK;
        #endif //GPSACTIVECHECK
        GPS_coord[LAT]=
        GPS_coord[LON]=
        GPS_altitude=
        GPS_fix_temp=
        GPS_numSat=
        GPS_speed=
        gpsvario();            
        if (GPS_fix_temp>0){
          GPS_fix=1;
        }
        GPS_NewData();
      }            
      GPS_ground_course=
      int16_t MwHeading360=GPS_ground_course/10;
      if (MwHeading360>180){
        MwHeading360 = MwHeading360-360;
        MwHeading   = MwHeading360;
      }

dustin, 2015-10-23:
Project update:  https://www.youtube.com/watch?v=_v3TA73zNkc

I've yet to have to do anything really special.  Well, except my special acro+ font mod:

[image]

ShikOfTheRa, 2015-10-23:
Looking good - that was quick!.....
Don't suppose you use ardupilot as well??! Always get asked about that. Probably do it in next release.

I am so glad I never post images of myself with goggles on testing...!!!! if the guys at work ever saw it would be all over facebook!

dustin, 2015-10-23:
I don't have any ardupilot at the moment.  I understand that's mavlink, though, which is also supported by TL.  The protocol is a good deal bigger than MSP from what I've seen.  It might be possible with a bit of refactoring to allow for more protocols to work here.

My initial TL goals were to bring TL close to MWOSD because the software is really quite nice as is.  There was other software out there for running mavlink on a minim type OSD.  I understand they're a lot slower and just... well, decrepit.  At least that's what the TL consensus seemed to be.

It might be fun to try to do some basic mavlink as well.  Also, some MSP extensions could be cool.  I'd like an unsolicited textual alert, for example.  :)

ShikOfTheRa, 2015-11-30:
How is this looking?

dustin, 2015-11-30:
It's great.  So far, I've added a `TAULABS` target define that is very similar to baseflight except it enables the alarm feature.  We've incorporated that into both Tau Labs and a derivative project.

The new alarm feature has been really awesome.  I don't currently disable "LOW VOLTS," but it becomes redundant as there's a "BATTERY" alarm that can be raised from the flight controller.  The FC also has a "TIME" alarm that is raised when it computes the expected consumption will result in fewer than `n` seconds of remaining flight time, as well as loads of other platform-specific alarms like "we've lost the transmitter" and "we're unable to autonomously follow the specified path." (both of those are somewhat abbreviated on the screen)

I think there's a little more work that can be done for presentation, but I'm quite happy with the results of the system.  You can see it in use [in this short flight](https://www.youtube.com/watch?v=4dFT7DA8cmA) of my [150mm](http://i.imgur.com/pewPbnu.jpg) quad.  Note that `EVENT` is flashing on the screen periodically which indicates some part of the event publish queue is not delivering in a timely manner.  This is a bug we wouldn't know about if we never saw it show up.

dustin, 2015-11-30:
It's probably fine to close this since the spirit of it is in effect.

closed 2015-11-30 by dustin

ShikOfTheRa, 2015-11-30:
OK cool - nice to have this added. Thanks!!

caseyjmorton, 2016-03-11:
Can you guys please provide a brief tutorial on GCS configuration and wiring in TL?

dustin, 2016-03-11:
There's not really much to write, I don't think.

Enable module.
Configure one of the ports (depends on your flight controller) as MSP.
Plug OSD into that port.

It should just work.

caseyjmorton, 2016-03-12:
Ok i didn't realize there was a separate MSP setting for the ports. I had
them set up as MAVLink thinking there was some sort of MAVLink to MSP
bridge I had to enable. As far as the wiring, would I wire it up just like
another UART with Rx and Tx crossed I'm assuming?

Casey

dustin, 2016-03-12:
Yeah, mavlink is another protocol for a different set of features.  MSP is
used for MWOSD.  Wiring is hopefully what you'd expect.

caseyjmorton, 2016-03-12:
Yup that solved the problem. Thanks again!

caseyjmorton, 2016-04-01:
One last question, I have a GPS fix, but i dont seem to be getting any GPS-based data.  Any ideas?

dustin, 2016-04-01:
The original code didn't have GPS support because I haven't stuck a GPS on
any of my aircraft in a long time.  The upcoming dRonin release (just hit
beta) does.

caseyjmorton, 2016-04-01:
No worries, I can help testing if you let me know how to get my hands on
the beta.

Casey

dustin, 2016-04-01:
Release is imminent, but we'd appreciate it, I'm sure.  We're just
finishing up the beta release notes in a different bug.  :)

cpusmith, 2016-04-01:
Casey, if you can see MSP Telemetry under modules then you have a version that supports MSP for MWOSD although the later releases may have better support.

You can get the latest NEXT here: http://jar.lyle.org:8080/job/dronin/Nodes=winx86/ which are essentially betas of the future releases which is Tanto. When I install them I do it to a different directory to keep them separate.