ShikOfTheRa/scarab-osd #130: DISPLAY_PR values > 44 degrees are incorrect

ikura, 2015-10-20:
I have enabled DISPLAY_PR on a micro minimOSD using dev1.5.1 with the latest betaflight.  So far so good...  The Pitch values work fine from -44 deg to 44 deg, and then at 45 degrees it suddenly jumps to 100.  From there it continues rising until it hits another 'step'.  By the time I'm pitched forward at 90deg the value is showing me ~324degrees.

Roll seems to have similar issues - when I'm almost fully upside down I'm seeing roll values of 1100-1200 degrees.

I had extended my AHI pitch / roll limits to 900 (90 degrees), not sure if that's related...

ikura, 2015-10-20:
Incidentally, the AHI disappears around these step values as well.

ShikOfTheRa, 2015-10-20:
OK, had a quick look and can't see anything obvious so either the code assumes the  data type from BF (and no-one has used this before to highlight), or BetaF is putting out different PR data to multiwii. 
Will not be able to look at this for a week or two, but will confirm ...

ikura, 2015-10-21:
I had a quick look through the code, and I've made it work for my setup - not sure whether this is correct for other targets.

Screen.ino: lines 270 & 280 both use itoa to convert the (absolute) angle into a string, but they're using Base 5??  Seems wrong to me, and after switching it to 10 I get the behaviour I'd expect.  Is there a reason Base 5 should be correct?

ShikOfTheRa, 2015-10-21:
Can't think why it should be base5. Probably a typo at some point.
Maybe that's why the limits on angles were ? 

Thanks for looking at - I'll update

ShikOfTheRa, 2015-10-21:
Updated ! - thanks

nitro20, 2016-03-07:
please tell me what i must do to see when my quadcopter is at more then 20deg pitch set as default..seems the line stops at 3/4 screen and after the default 20 degree..i just dont know enough about this stuff to make these simple changes..can some please tell how to do this.

i tryed to load 1.6 but i get error 1 in the arduino box. prolly something simple too..

nitro20, 2016-03-07:
ok so mine does this too..naze32, what do i do..gonna have to pacific.

ShikOfTheRa, 2016-03-07:
pacific?

nitro20, 2016-03-08:
i dont know much about this computer stuff, is there is a way to adjust the pitch degrees so that it reads what the craft is doing, i have that the same problem as ikura.(above). but mine is naze32 full version, and osd is r1.5.

ikura, 2016-03-08:
I think @nitro20 is asking for something different than the issue I raised.  It seems he's asking about the Artificial Horizon, and whether you can scale the degrees so that it stays on screen for more extreme pitch angles.  In practice it seemed to disappear off the top / bottom of the screen at around 25-30 degrees - in races we're often at 80-85 degrees!

I've switched over to Kiss FC now so I no longer have the awesomeness of MW_OSD available to me so I'm going to unsubscribe from this issue.

ShikOfTheRa, 2016-03-08:
Ah - I'll add that on the wish list - no reason why not - does mean the horizon will not line up, but guess that's not an issue in this scenario.

For KISS - I'd really welcome someone adding KISS support - I think it would be really easy....

nitro20, 2016-03-09:
yes and what ikura is saying to would be great...especially for beginner fpv.(maybe)...i changed the pitch default to 500 and this seems to let the AHL go higher on the screen, to about 40 degrees, then i have the pitch degree display turned on so i can see after the AHL disapears. but after 40 degs and when the line disappears the pitch degrees jumps to 100 then and onward..just as ikura sayed his was doing. but if i knew what he did to fix that problem, then i would ok with this set up as it is..i dont use much of the osd..AHL, voltage, ALT., and throttle position is all i want to use..i just dont know what ikura was talking about in his post about screen ion lines 270 280..

ShikOfTheRa, 2016-03-10:
1.6 working - excellent.
Scaling - its in the queue - no idea when will be implemented. No promises, but I'd like it in 1.7

closed 2016-03-10 by ShikOfTheRa