ShikOfTheRa/scarab-osd #131: Beta 1.5.1 does not include Mac GUI

JohnOCFII, 2015-10-21:
The Beta 1.5.1 zip doesn't include the Mac GUI app.  I downloaded the version from the released 1.5, and that worked fine.

ShikOfTheRa, 2015-10-21:
Yes - unfortunately I don't have a mac and the version produced by the compiler doesn't work so the beta tests don't come with it - sorry...

If you have a mac, you can install processing and run the source code - or create your own executable. Its very easy - very similar to arduino look/feel

JohnOCFII, 2015-10-21:
> Yes - unfortunately I don't have a mac and the version produced by the compiler doesn't work so the beta tests don't come with it - sorry...

No problem.  I'll take a look at building my own executable.    Thanks for the fast reply.

closed 2015-10-21 by JohnOCFII

andyshinn, 2015-11-22:
I can't find any documentation running it with processing. Can you point us to the sketch folder in the source?

nathantsoi, 2015-11-23:
open processing and open:

[image]