ShikOfTheRa/scarab-osd #134: GPS altitude in end of flight stats?

jaffadog, 2015-10-23:
I know it's possible to display either or both baro and gps altitude in the OSD.  It looks like the end of flight stats page displays maximum baro altitude.  Is it possible to display maximum gps altitude instead?  thanks!

ShikOfTheRa, 2015-10-23:
it would require some code changes for that. certainly possible

ShikOfTheRa, 2015-10-26:
As an fyi, if select FIXEDWING, the OSD uses GPS  instead of BARO / MAG. Can then re-enable the MAG and it will be close to what tyou want (unless you want to see both baro and GPS and same tiem... 
#define USEMAGHEADING             // Undefine this to use MAG for FW heading instead of GPS (requires controller with MAG sensor) 

possibly:
//#define DISABLEGPSALTITUDERESET   // Disables automatic reset of GPS Altitude to zero at arm for FC that already provide this functionality. 

Added to the wish list anyway. Its not too bad so probably will be done in next release or 2

ShikOfTheRa, 2015-11-30:
Closing as on teh wishlist.

closed 2015-11-30 by ShikOfTheRa