ShikOfTheRa/scarab-osd #136: Is anyone flying blackbox + scarab-osd?

nathantsoi, 2015-10-25:
Currently it seems this is not possible since the serial port switches from MSP to blackbox logging when armed.

It would be awesome if the OSD could auto-switch input from MSP to blackbox when the FC is armed.

Would anyone else use this feature?

ShikOfTheRa, 2015-10-25:
Sorry - no plans to support this config.
Can you use other serial for OSD instead

timotto, 2015-11-07:
It works on two different serial ports with cleanflight 1.10, blackbox at 250000bps, osd at 115200bps

dustin, 2015-11-24:
This is a configuration issue on your flight controller, not an MWOSD bug.  Trying to detect when protocol and speed change so you can pick out relevant bits from an obscure protocol at a different baud rate to pick out a few bits of information for display would be quite a challenge.

Is there a standard blackbox protocol spec?  Perhaps you could construct a PR to sniff that protocol and operate on relevant frames received.

closed 2015-11-30 by ShikOfTheRa