ShikOfTheRa/scarab-osd #137: Newbie at a loss as how to proceed

Hwurzburg, 2015-10-30:
Am a relative newbie to github....
I have made and tested some enhancements to the code, but I started from the release R1.5 not the master....have cloned the master and now dont know the best way to share them...
the changes were to allow 3s/4s autodetection for us miniquad flyers without having to change values...and to allow use of the battery status icon on mah (when used) instead of voltage...both features set by defs,,,details:
1. Fixed the map function for show battery evolution...its being used wrong and wont yield a even distribution of icons(google to find how to set the ranges correctly for its integer math)...also set the limits to more accurately reflect battery capacity based on cell voltage
2.if AUTODETECTNUMCELLS is defined, will autodetect 3s or 4s....gui values for cells, and alarm voltage will be used to calculate a cell level alarm....so its the same if 3s of if 4s is used....batt evolution icon uses this also, if enabled..
3. if BATICON4AMPHR is defined, and show bat status enabled, will place icon in front of mah and show percentage of alarm value from gui...in this case alarm flash is at 80% of that value...gui value should be total bat capacity

all my buds are loving this for our miniquad racers....

let me know...thanks

ShikOfTheRa, 2015-10-31:
Hi - thanks for contribution! I can't take a look at this at the moment as travelling - but will do when get back.
It sounds fine in principle.

ShikOfTheRa, 2015-11-30:
Please review and resubmit based upon latest master.
#1 may be OK - I remeber it was not quite right. I didnt have time too look at it properly.
#3 is of interest. Yes, I think use GUI alarm values

closed 2015-11-30 by ShikOfTheRa