ShikOfTheRa/scarab-osd #144: OSD + Bluetooth

Baatezu, 2015-11-20:
MWOSD 1.5 (minimOSD) can work on Flip32 (cleanglight 1.10.0) like show on this pictures.
Waht need to commit on conf.h?
Or it's not possible?

nathantsoi, 2015-11-20:
I've never been able to get this to work. I think the bluetooth module and the OSD are both trying to send MultiWii serial protocol commands. If we could disconnect the MWOSD TX line, it might work, but I think this would make the OSD inoperable, since it must request the data it wants. Perhaps we could change CleanFlight to auto-send OSD info on a given port or change MWOSD to accept blackbox data :)

lichtl, 2015-11-20:
I think as stated in there you need to have either the Bluetooth TX or OSD TX being disconnected. I'm not 100% sure but I think the OSD should even work with only RX (however I believe it then won't allow you to perform pid/config changes using the OSD

Baatezu, 2015-11-20:
When a disconnect TX on OSD, OSD say NO DATA.

Try later recomendation @nathantsoi

dustin, 2015-11-24:
@lichtl the OSD requests data from the flight controller to display.  The flight controller doesn't just broadcast its state..

I don't see an MWOSD bug here.

tracernz, 2015-11-24:
@dustin It can using MSP telemetry, but looks like CF will remove that anyway cleanflight/cleanflight#1106.

dustin, 2015-11-24:
I'd assume you could share the (FC-side) tx pin and have it somewhat work if bluetooth is actually asking for stuff.  Seems like a bunch of work, though.  I guess a hardware MSP mux would also do it.  But just putting it on a different UART is the easiest approach.

ShikOfTheRa, 2015-11-30:
What is said above should cover everything

closed 2015-11-30 by ShikOfTheRa