ShikOfTheRa/scarab-osd #149: Waiting OSD

Zadrema, 2015-11-30:
Hi
I did flash a version 1.5 - everything worked very well, In FatShark I have seen information from OSDoge!
Today after run my QAV I did not see on screen messages from OSDoge. I connected OSDoge to my computer and after connect COM port, I view message:"Waiting OSD..."
I read this information - 21 I get "Waiting FC” or "Waiting OSD". How do I fix it.
-Ensure you are connected to the correct com port. - OK
-Ensure baud rate in GUI matches baud rate on the OSD - OK
-Ensure only the OSD is connected to programmer - OK
-If using OSDOGE ensure switch is in correct position for programming - OK
-If using cheap OSDOGE/BROSD clone, disconnect board from the Naze32 - OSDoge)
-Restart PC – especially if hibernated - OK
-Ensure physical connections are correct. Test by re-uploading MWOSD if necessary to verify - OK
-Upload EEPROM_Clear example sketch then re-upload MWOSD sketch. - ERROR
-Set serial port default value on PC to be same as OSD (e.g. device manager>ports>comx>port settings) -OK

Next I used arduino 1.6.6 with default Config.h for "re-upload MWOSD sketch" (MWOSD 1.5) :
Arduino: 1.6.6 (Windows 10), "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"
...
Using Port : COM12
Using Programmer : arduino
Overriding Baud Rate : 57600
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x9f
....
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x9f
avrdude done. Thank you.
Trouble downloading. Help on downloading: http://www.arduino.cc/en/Guide/Troubleshooting#upload. 

Please tell me how can I restore my OSD?

ShikOfTheRa, 2015-12-01:
Have you disconnected the OSD from naze?

If using cheap OSDOGE/BROSD clone, disconnect board from the Naze32 - OSDoge)

Zadrema, 2015-12-01:
Yes, I  disconnected the OSD from naze. Right position port on the OSDoge and only USB connect.
LEDs light - constant one blue. 
PS:similar computer in which no HDD )))

ShikOfTheRa, 2015-12-02:
If you are having problems uploading the sketch - avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x9f

Then this is not MWOSD - it usually means a problem with wiring or fault with board

Zadrema, 2015-12-02:
OK. Thanks you.

closed 2015-12-02 by Zadrema