ShikOfTheRa/scarab-osd #155: Batt status icon should be optionally used for total MAH used in systems with current measuring

Hwurzburg, 2015-12-01:
battery capacity and alarms are better than min voltage in high amperage uses like miniquads for judging low battery due to high voltage drops under heavy loads...see my previous pull request for this enhancement

ShikOfTheRa, 2015-12-02:
Yes - will have a look. Its nice functionality for those that use it. Mucho better than voltage.

ShikOfTheRa, 2015-12-03:
OK cut and pasted in. Works great!

What do you think of this...
In config.h you choose battery "evo" icon.  used is voltage, current - or both.... 
Voltage enabled by default as I think that 95% use that only.
GUI switch enables or disables whatever choose in the config.h

Hwurzburg, 2015-12-03:
not sure about both....might be contradictory....I would think volt OR current evo icon

ShikOfTheRa, 2015-12-03:
I bet someone will want it sooner or later....  
Possible scenarios of using different battery sizes / not fully charged.

One for the todo list. People can use what they want then.
Personally I don't normally use anything other than volts and timer...

ShikOfTheRa, 2015-12-04:
OK done anyway as a quickie.

Again - thanks for the contribution. HAve a current sensor on my plane so will try it out once able to fly again.

closed 2015-12-04 by ShikOfTheRa