ShikOfTheRa/scarab-osd #156: Inconsistent font changes 1.5.0 to 1.5.1

flacjacket, 2015-12-01:
I noticed big changes between the fonts and the distinction between bold vs large vs large high contrast. This may be intentional, but I figured I'd bring it up.

This summarizes the changes:
[image]

Things like the current "bold" numbers, which look like the "large" numbers seem a bit out of place, and the current "large high contrast" looks more like a "bold" font set on the whole.

It seems like things got a bit jumbled in 7eca29e7a502267be15e5b726cdfe5249a61245c, since that commit message is "Crosshair fix for large font", but it in fact changes the bold fonts. Maybe the current "bold" should be "large", the current "large" should be "large high contrast" and the current "large high contrast" should be "bold"?

ShikOfTheRa, 2015-12-02:
Oh yes. Thanks for that. I dont use them myself. Not quite sure what happened there. Think I'll drop one of them as well - the differences between 2 of them a re so small its not worth maitaining

ShikOfTheRa, 2015-12-02:
OK hopefully done correct now - thanks!

closed 2015-12-02 by ShikOfTheRa