ShikOfTheRa/scarab-osd #157: GUI not working on Mac OSX

DTFUHF, 2015-12-02:
Unfortunately I don't have a Mac to try this on myself. I've received a report of 1.5 and older GUIs not working on mac. Quote from the user:

the file is damaged and suggests they should be deleted.

Version 1.2 is working for this user. Additionally, 1.5.1.0 release does not include any Mac version.

edgarasm, 2015-12-02:
They have to select "Allow app downloaded from: Anywhere" in security & privacy settings, at least for the first time

DTFUHF, 2015-12-03:
Thanks very much, this has fixed the issue for the user. Sounds like the type of thing I'd immediately disable :)

I see this is already in the FAQ, so I'll close this.

closed 2015-12-03 by DTFUHF

tracernz, 2015-12-03:
When this happens, you can also usually launch the app, it fails, go to Security & Privacy, click Open Anyway and it will be fine from then on.