ShikOfTheRa/scarab-osd #163: Mac app doesn't seem to work on Release 1.5.2 (Pre-1.6)

basdelfos, 2015-12-13:
Pulled the latest version from the master repository today with release 1.5.2 (pre-1.6). The Mac GUI app doesn't seem to work in this release. It crashes at startup.

Logs:
`13/12/15 20:40:16,839 com.apple.xpc.launchd[1]: (MW_OSD_GUI.183392[2295]) Service exited with abnormal code: 1`

`13/12/15 20:40:38,638 com.apple.xpc.launchd[1]: (com.apple.ReportCrash.Root[2307]) Endpoint has been activated through legacy launch(3) APIs. Please switch to XPC or bootstrap_check_in(): com.apple.ReportCrash.DirectoryService`

nathantsoi, 2015-12-13:
did you try grabbing the .app from the latest code or from the release?

try downloading the code, instead of the release:
https://github.com/ShikOfTheRa/scarab-osd/archive/master.zip

then open MW_OSD_GUI/application.macosx/MW_OSD_GUI.app

basdelfos, 2015-12-13:
I did a fresh pull from the master branch and then opened MW_OSD_GUI/application.macosx/MW_OSD_GUI.app. 
The same problem still occurs when i download the zip file in the link you supplied.

I just noticed that there is a huge difference in file size of MW_OSD_GUI.app between R1.5 and pre-R1.6. In R1.5 release the .app file size is 6.3MB and in the pre-R1.6 release .app file size is 100.2MB! Probably something is messed up with the simlinks.

nathantsoi, 2015-12-13:
Hum... maybe i broke it... can you try running the app via processing as
described here?
https://raw.githubusercontent.com/ShikOfTheRa/scarab-osd/master/OTHER/MW_OSD_GUI_SOURCE/MW_OSD_GUI/setup.sh

nathantsoi, 2015-12-13:
also, are you allowing apps from anywhere?

[image]

ps. the app is bigger b/c i embedded the jdk, so it should work more consistently across different computers: https://github.com/processing/processing/wiki/Export-Info-and-Tips

basdelfos, 2015-12-13:
Yes, allow apps downloaded from anywhere is configured already.
I will try running the app via processing tomorrow, the script uses wget which is not available on El Capitan (maybe older versions too). I tried to use curl, but curl seems to only download the first 600 bytes of the processing app. I have to figure that out or use another download command.

To be continued...

nathantsoi, 2015-12-13:
ahh, probably need to `brew install wget` after installing http://brew.sh/, ill change that to use curl (or you can just do it manually)

basdelfos, 2015-12-14:
I have installed the processing app via the setup script. I had to change the script a bit for using curl. I will make a push request for this.

Good news is that the Mac app seems to work when I run it within the processing app. 
The sourcecode is not causing the GUI app te crash, something seems to be wrong in the build of the GUI app.

nathantsoi, 2015-12-14:
weird can you try exporting the mac app in processing and emailing it to
me? i can try it on my mac

basdelfos, 2015-12-14:
I will do. I used these settings to export the app:

[image]

Source:
./OTHER/MS_OSD_GUI_SOURCE/MW_OSD_GUI/MW_OSD_GUI.pde

nathantsoi, 2015-12-16:
seems like your pull may have fixed this?

ShikOfTheRa, 2015-12-16:
You guys will have to advise. Not a mac guy!!1

basdelfos, 2015-12-16:
Yes, it seems to be solved by generating the Mac app. The generated app that I made also works on nathantsio's Mac.

But the strange thing is that I just downloaded a new zip of the master branche and the compiled Mac app is not working again. This is even more strange because the this Mac app is from a pull request from my fork, and it worked yesterday. I think something get messed up by GitHub...

JohnOCFII, 2015-12-19:
I just downloaded from Master, and see the same abnormal exit.  Trying on El Capitan.

12/19/15 3:42:22.490 PM com.apple.xpc.launchd[1]: (MW_OSD_GUI.264992[7820]) Service exited with abnormal code: 1

Anything else I can try to help troubleshoot?

John

nathantsoi, 2015-12-21:
Does running from source work?

https://raw.githubusercontent.com/ShikOfTheRa/scarab-osd/master/OTHER/MW_OSD_GUI_SOURCE/MW_OSD_GUI/setup.sh

JohnOCFII, 2015-12-21:
> Does running from source work?

I'll give that a shot later tonight (in about 5 hours).

JohnOCFII, 2015-12-22:
I must be missing something.  I downloaded and installed the Processing App and the library.  From the Processing App I pointed it at the MW_OSD_GUI.pde file.  It wanted it in a different directory, so I let it created the MW_OSD directory and move the file.  After that, I got some errors in the sketch, so I'm assuming I'm still missing something.

Right now I've got the following directory structure:  

/Users/osh/Documents/RC Stuff/MinimOSD/scarab-osd-master/MW_OSD_GUI/application.macosx/source/MW_OSD_GUI

Do I need to move that directory under /Users/osh/Documents/Processing/libraries ?

Thanks,

basdelfos, 2015-12-22:
@johnOCFII - did you download version 2.2.1 of the processing app? If you have downloaded the newest version, you will get compile errors due to syntax changes.

You can use this script to automatically download and install the processing app and library. 
https://raw.githubusercontent.com/ShikOfTheRa/scarab-osd/master/OTHER/MW_OSD_GUI_SOURCE/MW_OSD_GUI/setup.sh

JohnOCFII, 2015-12-22:
@basdelfos Yes - I did use the script and downloaded version 2.2.1  of the Processing App.  I can successfully load and run example applications.  When I attempt to run /Users/osh/Documents/RC Stuff/MinimOSD/scarab-osd-master/MW_OSD_GUI/application.macosx/source/MW_OSD_GUI/MW_OSD_GUI.pde I get null pointer exceptions (as below) leading me to believe I don't have all the MWOSD_GUI files in the right place for the Processing App to find them.

>hudlayout.xml does not exist or could not be read
>java.lang.NullPointerException
>	at processing.data.XML$1.read(XML.java:190)
>	at com.sun.org.apache.xerces.internal.impl.XMLEntityScanner.load(XMLEntityScanner.java:1753)
>	at com.sun.org.apache.xerces.internal.impl.XMLEntityScanner.arrangeCapacity(XMLEntityScanner.java:1629)
>	at com.sun.org.apache.xerces.internal.impl.XMLEntityScanner.skipString(XMLEntityScanner.java:1667)
>	at com.sun.org.apache.xerces.internal.impl.XMLVersionDetector.determineDocVersion(XMLVersionDetector.java:196)
>	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:812)
>	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:777)
>	at com.sun.org.apache.xerces.internal.parsers.XMLParser.parse(XMLParser.java:141)
>	at com.sun.org.apache.xerces.internal.parsers.DOMParser.parse(DOMParser.java:243)
>	at com.sun.org.apache.xerces.internal.jaxp.DocumentBuilderImpl.parse(DocumentBuilderImpl.java:347)
>	at processing.data.XML.<init>(XML.java:187)
>	at processing.core.PApplet.loadXML(PApplet.java:6322)
>	at processing.core.PApplet.loadXML(PApplet.java:6312)
>	at MW_OSD_GUI.setup(MW_OSD_GUI.java:782)
>	at processing.core.PApplet.handleDraw(PApplet.java:2361)
>	at processing.core.PGraphicsJava2D.requestDraw(PGraphicsJava2D.java:240)
>	at processing.core.PApplet.run(PApplet.java:2256)
>	at java.lang.Thread.run(Thread.java:745)
>processing.app.SketchException: NullPointerException
>	at processing.mode.java.JavaBuild.placeException(JavaBuild.java:713)
>	at processing.mode.java.runner.Runner.findException(Runner.java:896)
>	at processing.mode.java.runner.Runner.reportException(Runner.java:871)
>	at processing.mode.java.runner.Runner.exceptionEvent(Runner.java:797)
>	at processing.mode.java.runner.Runner$2.run(Runner.java:688)

basdelfos, 2015-12-23:
@JohnOCFII - I see now that you are using the source from the wrong directory. You need to run the application from the following location: 
scarab-osd/OTHER/MW_OSD_GUI_SOURCE/MW_OSD_GUI/MW_OSD_GUI.pde

This location has all the files to build and run the application within the Processing app.

Don't ask me what the source folder is doing in the same folder as the compiled app, maybe someone else knows why...

JohnOCFII, 2015-12-23:
@basdelfos Thanks!  That did the trick.  I was able to run the App from within the Processing environment, as well as use the export capability to create a standalone App.  I created and tested both with and without the included Java JDK.  I also copied the version with Java to a different Mac, and was able to successfully launch the App from that Mac too.

So - yes, it seems something is broken with packaging on GitHub.

John

> You need to run the application from the following location: 
> scarab-osd/OTHER/MW_OSD_GUI_SOURCE/MW_OSD_GUI/MW_OSD_GUI.pde
> 
> This location has all the files to build and run the application within the Processing app.

ShikOfTheRa, 2015-12-24:
"Don't ask me what the source folder is doing in the same folder as the compiled app, maybe someone else knows why..."

Processing includes source with app when it compiles. Not sure why either.

ShikOfTheRa, 2016-01-06:
Think we can close this now?!

closed 2016-01-06 by ShikOfTheRa

basdelfos, 2016-01-06:
Yes, issue can be closed now.