ShikOfTheRa/scarab-osd #169: Wrong amps from Naze32

pawelb07, 2016-01-04:
I get about 10 times higher amps then what is shown in cleanflight (Naze32) and Taranis. I use Frsky D4R-II receiver and a current sensor which is hoked up to Naze32 (Pin 8). The current sensor is well calibrated in Naze32 (50mV/A).
I tried many options and both firmwares: 1.5 and the newest 1.5.2 and I still can't get read of this issue.

axyspg, 2016-01-04:
I wonder if has something to do with the "Enable support for legacy Multiwii MSP current output" on Cleanflight configurator "Configuration" tab, with a fast check in the Cleanflight code I can see amperage multiplied by 10. It's just a blind guess, I don't have current sensor so can't know for sure.

```
        if(masterConfig.batteryConfig.multiwiiCurrentMeterOutput) {
            serialize16((uint16_t)constrain(amperage * 10, 0, 0xFFFF)); // send amperage in 0.001 A steps. Negative range is truncated to zero
        } else
            serialize16((int16_t)constrain(amperage, -0x8000, 0x7FFF)); // send amperage in 0.01 A steps, range is -320A to 320A
        break;
```

pawelb07, 2016-01-04:
axyspg - a good point, but I already checked this possibility. With enabled "  support for legacy Multiwii MSP current output" the amp value on OSD is x100. 
x10 is when the option is disabled.
The funny thing is that even with this option enabled in cleanflight, my Taranis shows  amps correctly. Off course in this case the value in cleanflight is x10.

axyspg, 2016-01-05:
I see... Sorry but as I don't have that sensor can't be of much help. Just guessing again blindly, I suppose you can workaround this problem editing Def.h definition of Amperage Divider, setting a 100 value instead of 10. Not sure if this is the best way, just guessing again...

value definition in Def.h is like this:
```
#if defined (CLEANFLIGHT190)
  #define AMPERAGE_DIV 10
#endif
```

ShikOfTheRa, 2016-01-05:
Yes can manipulate in def.h as axyspg advises, It may be OSD code is incorrect. There have been changes, but as far as aware have had no issues reported.

but really need to find out what is wrong here. Had a quick look through the code on CF and OSD and the logic seems to match up.  
CF sens 100A as 10,000 when using defaults.
OSD  divides this by 10 to get 1,000
Then displays the 1000 as 100.0A 
this is using CLEANFLIGHT define, If use default MULTIWII it will be 10* higher...

pawelb07, 2016-01-05:
Thanks guys for your hints!
I haven't had much time today to check all the options, but I tried to play with amperage divider in def.h file. Unfortunately setting 100 value instead of 10 didn't work. What is strange the value presented on the OSD screen was the same as with 10 value?
Then I bricked my OSD perhaps by trying to change too many things in the code :-)
Well, in fact it's not bricked - it works, but now I can't clear it in order to write any other firmware.I also can't connect through Windwos GUI software, although I can program it through my TX. It's micro minim OSD board. Could you give me any hint in this matter? I have no idea what is a reset button alternative in this board? I can't burn a new bootloader too.

axyspg, 2016-01-05:
May be cleanflight is not defined in the config.h file. Make sure you have cleanflight defined instead of multiwii. 

If I understand correctly you can't flash because no COM port is detected. In this case check USB cable is OK and the rs232 dongle is working and drivers are there.

I've found a tutorial to flash, I think it's not exactly your case, but just in case is helpful 

https://quadmeup.shtr.eu/how-to-flash-mw-osd-to-micro-minimosd/

pawelb07, 2016-01-05:
Bingo - I have defined multiwii instead cleanflight. Unfortunately, I can't check it right now...

Well, com port is detected, but it looks like the board is write protected now. When i use FTDI - the red led blink only few times when I try to write something and then only green led is blinking. The writing process is never ending until I unplug the board. Then off course I can see the error message in the arduino. I need to check yet with other RS232, but I think it's not the case. I'm affraid I need to buy next minimosd.

axyspg, 2016-01-05:
Looks like the board doesn't autoreset. Make sure DTR cable is correctly connected, I think that resets the board to enter I to the bootloader.

pawelb07, 2016-01-05:
I resoldered evrything already, but maybe there's something wrong with the FTDI board. I will try to borrow one from my friend tomorrow and who knows..

ShikOfTheRa, 2016-01-06:
OK - think we have the answer now.

For support can you use the fpvlab thread - a whole bunch more of people available to help there - and quicker and I would like to kepe github purely for dev stuff if possible - thanks!!

closed 2016-01-06 by ShikOfTheRa